MarlinFirmware/Marlin #23578: [FR] be able to set FIL_RUNOUT_INVERTING through gcode or UI

Diz-X, 2022-01-20:
### Is your feature request related to a problem? Please describe.

I want to use a filament runout sensor with inverted status, problem is I need to compile the firmware. 

### Are you looking for hardware support?

Ender 3 v2 board version 4.2.2 

### Describe the feature you want

A way to set FIL_RUNOUT_INVERTING in the UI or with gcode, without the need to compile the code yourself.

DejitaruJin, 2022-01-20:
Can I ask what you're needing to accomplish with this? From the electrical standpoint, there are actually two important #defines for that, `FIL_RUNOUT_STATE` and either `FIL_RUNOUT_PULLUP` or `FIL_RUNOUT_PULLDOWN` (or neither). Since these values need to match the specific switch you're using, they definitely need to be hard-coded at compile time. Is there a reason you'd need to swap the value while the machine is running?

Diz-X, 2022-01-21:
The reason I ask is to make it more user friendly for beginners. As there are a lot of sensors on the market, non are created equaly and a lot of people don't know how to compile the firmware from source. I understand that this is adding a extra layer to the code, but could help a lot of headache for people with reversed sensors.

DejitaruJin, 2022-01-21:
I fear that might only move the headache elsewhere. A beginner who is replacing a runout sensor should replace it with one known to work with that model of printer, in which case the existing firmware for it will work already. _Adding_ a runout sensor probably isn't something a beginner should take on - and it requires recompiling the code regardless, because `#define FILAMENT_RUNOUT_SENSOR` is commented out by default, and even `M412` won't turn it on because the code was compiled without support entirely. And then you have to consider which pin the sensor is actually hooked up to, which varies by model...

Diz-X, 2022-01-25:
I can understand, because of small group of people like me that use custom switches this could be confusing for others. 

I have opened up the sensor and resoldered/rewired the switch, now the default pre-compiled firmware correctly. I am way too used writing software that is configurable in every sense of the way. Firmware is another beast on its own.

closed 2022-01-25 by Diz-X