MarlinFirmware/Marlin #23579: [BUG] (bug summary) Firmware not compiled with USB flash drive support for BTT SKR PRO v1.2

svd05, 2022-01-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Firmware not compiled with USB flash drive support for BTT SKR PRO v1.2
[image]
[image]

### Bug Timeline

It's new problem. Compiling first time with USB support.

### Expected behavior

Firmware will be compiled.

### Actual behavior

Errors. Firmware not compiled.

### Steps to Reproduce

1. Configure "Configuration.h, Configuration_adv.h, pins_BTT_SKR_PRO_common.h, platformio.ini" with USB support.
2. Compiling.

### Version of Marlin Firmware

2.0.9.3, 2.0.x-bugfix

### Printer model

own

### Electronics

BTT SKR Pro v1.2

### Add-ons

MAX31865, PT100

### Your Slicer

Cura

### Host Software

Cura

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/7906879/config.zip)

Bob-the-Kuhn, 2022-02-02:
I was not able to compile using any comination of  USE_UHS2_USB vs. USE_UHS3_USB and your platformio.ini vs. the default.

None of my error messages matched yours.

Are you using an article to guide you through adding USB flash drive support?  If yes, please include a link to the article.

I'm concerned that the Marlin's USB flash drive support doesn't pertain to your hardware.  The configuration_adv.h file says it is for MAX3421E breakout board solutions.  While the BTT SKR PRO does have two independent USB controllersw built into the CPU, I don't think that Marlin currently can make use of both.  Hope I'm wrong on this.

Bob-the-Kuhn, 2022-02-04:
I don't know enough about the USB FLASH drive option to help on this issue.

I did finally notice the comment indicating that the OTG port on boards like yours may be supported.

I also finally noticed the **BIGTREE_SKR_PRO_usb_flash_drive** build option.

I have not been able to compile using BIGTREE_SKR_PRO_usb_flash_drive.  I've tried the following:

- `//#define USE_UHS2_USB` with  `#define USE_UHS3_USB`
- `#define USE_UHS2_USB` with  `//#define USE_UHS3_USB`
- `//#define USE_UHS2_USB` with  `//#define USE_UHS3_USB`
- `//#define USE_UHS2_USB` with  `//#define USE_UHS3_USB` and serial port 2 disabled

thinkyhead, 2022-04-10:
First, it looks like there's no Native USB Serial for this board, so maybe this should be added to the pins file…
```cpp
#define BOARD_NO_NATIVE_USB
```
Beyond that, we can only compare with other boards that successfully compile with `USE_UHS3_USB` and `USE_OTG_USB_HOST` both enabled. In my testing with your configurations I could only get a successful compile with `USE_UHS2_USB` enabled and `USE_OTG_USB_HOST` disabled.

svd05, 2022-04-27:
> USE_UHS2_USB enabled and USE_OTG_USB_HOST disabled

tried and got this:
[image]
[image]
[image]
etc.

wcdnail, 2022-10-19:
[image]
Change SERIAL_PORT to 1

wcdnail, 2022-10-19:
[image]

USE_UHS2_USB & USE_UHS3_USB must be disabled
USE_OTG_USB_HOST must be enabled

On SKR Pro 1.1 board I've got detection issue, but M21 g-code works correctly

svd05, 2023-03-07:
better than it was, but there are still errors
[image]

ellensp, 2023-03-08:
@svd05 you did not build with the required build environment BIGTREE_SKR_PRO_usb_flash_drive

svd05, 2023-03-09:
@ellensp, you're right, but this is no changing result.
[image]

thisiskeithb, 2023-05-03:
> First, it looks like there's no Native USB Serial for this board, so maybe this should be added to the pins file…
> 
> ```c++
> #define BOARD_NO_NATIVE_USB
> ```

It does, but not when using `default_envs = BIGTREE_SKR_PRO_usb_flash_drive` / `USB_FLASH_DRIVE_SUPPORT`. `#define SERIAL_PORT -1` / usb serial only works with `default_envs BIGTREE_SKR_PRO`.

You also have to physically move two jumpers to select between the USB-B (serial/host) port or USB-A (flash drive) port:
[image]

I spent some time porting OP's config to the latest bugfix-2.1.x (dbed3f1) and after fixing the serial ports and disabling `USE_UHS3_USB`, it compiles and boots. Flash drives can be read after moving the onboard jumpers over.

Here's the compiled firmware for easy testing: [firmware.bin.zip](https://github.com/MarlinFirmware/Marlin/files/11378860/firmware.bin.zip)

Here are the changed files: [issue_235789_changed_files_bugfix-2.1.x_commit_dbed3f1.zip](https://github.com/MarlinFirmware/Marlin/files/11378869/issue_235789_changed_files_bugfix-2.1.x_commit_dbed3f1.zip)

<details><summary>And a diff for good measure:</summary>
<p>

```diff
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 8818b04..185cea3 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -67,7 +67,7 @@
 
 // Choose the name from boards.h that matches your setup
 #ifndef MOTHERBOARD
-  #define MOTHERBOARD BOARD_RAMPS_14_EFB
+  #define MOTHERBOARD BOARD_BTT_SKR_PRO_V1_2
 #endif
 
 /**
@@ -78,7 +78,7 @@
  *
  * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
  */
-#define SERIAL_PORT 0
+#define SERIAL_PORT 1
 
 /**
  * Serial Port Baud Rate
@@ -115,7 +115,7 @@
 //#define BLUETOOTH
 
 // Name displayed in the LCD "Ready" message and Info menu
-//#define CUSTOM_MACHINE_NAME "3D Printer"
+#define CUSTOM_MACHINE_NAME "3D Printer"
 
 // Printer's unique ID, used by some programs to differentiate between machines.
 // Choose your own or use a service like https://www.uuidgenerator.net/version4
@@ -138,9 +138,9 @@
  *          TMC5130, TMC5130_STANDALONE, TMC5160, TMC5160_STANDALONE
  * :['A4988', 'A5984', 'DRV8825', 'LV8729', 'TB6560', 'TB6600', 'TMC2100', 'TMC2130', 'TMC2130_STANDALONE', 'TMC2160', 'TMC2160_STANDALONE', 'TMC2208', 'TMC2208_STANDALONE', 'TMC2209', 'TMC2209_STANDALONE', 'TMC26X', 'TMC26X_STANDALONE', 'TMC2660', 'TMC2660_STANDALONE', 'TMC5130', 'TMC5130_STANDALONE', 'TMC5160', 'TMC5160_STANDALONE']
  */
-#define X_DRIVER_TYPE  A4988
-#define Y_DRIVER_TYPE  A4988
-#define Z_DRIVER_TYPE  A4988
+#define X_DRIVER_TYPE  TMC2209
+#define Y_DRIVER_TYPE  TMC2209
+#define Z_DRIVER_TYPE  TMC2209
 //#define X2_DRIVER_TYPE A4988
 //#define Y2_DRIVER_TYPE A4988
 //#define Z2_DRIVER_TYPE A4988
@@ -152,7 +152,7 @@
 //#define U_DRIVER_TYPE  A4988
 //#define V_DRIVER_TYPE  A4988
 //#define W_DRIVER_TYPE  A4988
-#define E0_DRIVER_TYPE A4988
[... 734 lines cut ...]
+  #define TEMP_0_SCK_PIN                    PC10
+  #define TEMP_0_MOSI_PIN                   PC12
+  #define TEMP_0_CS_PIN                     PG12
+#endif
+
 //
 // Heaters
 //
diff --git a/platformio.ini b/platformio.ini
index e3bdb6f..0f27dc4 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -13,7 +13,7 @@
 [platformio]
 src_dir      = Marlin
 boards_dir   = buildroot/share/PlatformIO/boards
-default_envs = mega2560
+default_envs = BIGTREE_SKR_PRO_usb_flash_drive
 include_dir  = Marlin
 extra_configs =
     Marlin/config.ini
```

</p>
</details>

closed 2023-05-03 by thisiskeithb