MarlinFirmware/Marlin #23580: [BUG] RUMBA NeoPixel pin wrong

kab01m, 2022-01-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Neopixel LCD backlight does not work on RUMBA board. Tested on MINI12864 V3.0. The problem is NEOPIXEL_PIN in Marlin/src/pins/ramps/pins_RUMBA.h which is 25, but schematic says it is 38 (50 pin, D7). Got my screen working with this diff:
`
--- Marlin-2.0.x/Marlin/src/pins/ramps/pins_RUMBA.h.orig	2022-01-21 03:17:50.000000000 +0300
+++ ./Marlin-2.0.x/Marlin/src/pins/ramps/pins_RUMBA.h	2022-01-21 15:29:10.935155231 +0300
@@ -221,5 +221,5 @@
     #endif
   #elif ENABLED(FYSETC_MINI_12864_2_1)
-    #define NEOPIXEL_PIN                      25
+    #define NEOPIXEL_PIN                      38
   #endif
`

### Version of Marlin Firmware

2.0.9.3

### Printer model

Kossel Delta Custom

### Electronics

RUMBA

Bob-the-Kuhn, 2022-01-29:
Looks like you found an error in my PR #13900.  Srange no one has found it in the years since.

25 is definitely wrong.  That's the E1_DIR line.

I'll start a PR to fix this.

Bob-the-Kuhn, 2022-01-29:
PR #23646 fixes this issue.

closed 2022-01-30 by thisiskeithb