MarlinFirmware/Marlin #23581: [BUG] Frequent TEMP MALFUNCTION errors on certain Archim based printers

marciot, 2022-01-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The thermal protection feature introduced in https://github.com/MarlinFirmware/Marlin/pull/23373 seems to cause frequent temp malfunction errors on some printers based on the Archim board. This was first observed on an unreleased SynDaver prototype, but has now also been independently reported by a user who upgraded a LulzBot TAZ 6 printer to the Archim using Drunken Octopus firmware.

There doesn't appear to be any way in the config files to tune the sensitivity of this feature, short of disabling it entirely. Looking for suggestions.

### Bug Timeline

This started once the code was upgraded to include this change.

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Unreleased SynDaver prototype; Lulzbot TAZ 6

### Electronics

Ultimachine Archim

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

Other (explain below)

### Additional information & file uploads

```
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Error:Thermal Malfunction, system stopped! Heater_ID: E0

echo:busy: processing
echo:busy: processing
echo:busy: processing
X:10.00 Y:183.00 Z:20.00 E:179.04 Count X:1000 Y:18300 Z:60000
echo:TEMP MALFUNCTION
Error:Printer halted. kill() called!
//action:poweroff
```

thisiskeithb, 2022-01-21:
Can you try https://github.com/MarlinFirmware/Marlin/pull/23571?

CC: @zeleps

marciot, 2022-01-21:
> Can you try #23571?
> 
> CC: @zeleps

Applying the patch, and uncommenting the override which effectively increases the period from 20 seconds to 120 seconds, does seem to help on a printer that was having a problem before. I have not seen the error yet, but it may only be a matter of waiting long enough for it.

zeleps, 2022-01-21:
@marciot 20'' is obviously a very small time window for a stable heater, but it's not entirely impossible that you may have a stable temperature for 120''. If you eventually do get a thermal malfunction with this setting, make sure it's not a real malfunction (reset your printer asap and see if the temperature reading is actually higher than expected), and if it isn't I suggest you turn the feature off for the shake of stability.

marciot, 2022-01-26:
@zeleps: It seemed good for about six hours sitting idle, I did not test it beyond this time period.

johan2406, 2022-02-06:
Hey all. I'm having the same issue with my Ender 3 Pro with upgrade board V4.2.7. I even replaced the thermistor on the bed and did a 3 hour print without an issue. Loaded my next print, about 20min in it gave the same error.

 Error:Thermal Malfunction, system stopped! Heater_ID: bed

Checked and rechecked al the cabling. Tested with a multimeter and everything is 100%

Any advice?

thisiskeithb, 2022-02-06:
@johan2406 Disable `THERMAL_PROTECTION_VARIANCE_MONITOR` for now. It tends to give false positives on stable systems.

Funny62, 2022-02-23:
Same issue on Ender 3 S1, and fw mriscoc last FW...
https://github.com/mriscoc/Ender3V2S1/releases/tag/20220220

thisiskeithb, 2022-02-23:
> Same issue on Ender 3 S1, and fw mriscoc last FW...

My same advice applies to disable the feature for now.

Also, we don’t support forks of Marlin since we have no control over what they do, not to mention that mriscoc’s firmware is closed source, so there is no telling what else is changed.

grrmisfit, 2022-02-24:
problem with this issue I have is I have 2 S1s in same room running exact same firmware, one gets the temp issues the other does not.

thinkyhead, 2026-10-08:
Thanks for reporting this. A lot has changed in this area since it was filed. Could you please flash the current `bugfix-2.1.x` and tell us whether the problem still happens? If it does, please attach your current `Configuration.h` and `Configuration_adv.h` and a short description of the steps. We're trying to clear the decks for the 2.1.3 release, so a quick retest helps a lot.

closed 2026-10-20 by thinkyhead