MarlinFirmware/Marlin #23582: [BUG] 2.0.9.3 Ender 3V2 UBL mess is missing after reboot

billygr, 2022-01-21:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

UBL mesh is not available after reboot, although it is saved correctly

I am using UBL 10x10 config files attached

### Steps to Reproduce

Step to reproduce it:
M502            ; Reset settings to configuration defaults...
M500            ; ...and Save to EEPROM. Use this on a new install.
M501            ; Read back in the saved EEPROM.
G28       ; home all axes
M155 S30  ; reduce temperature reporting rate to reduce output pollution
M190 S65  ; (optional) wait for the bed to get up to temperature
G29 P1    ; automatically populate mesh with all reachable points
G29 P3    ; infer the rest of the mesh values
G29 P3    ; infer the rest of the mesh values again

M420 S1 V ; enabled leveling and report the new mesh
Send: M420 S1 V
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     ( 10,220)                                                              (220,220)
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:  9 | +0.008  -0.000  -0.009  -0.017  -0.025  -0.030  -0.041  -0.059  -0.070  -0.070
Recv:    |
Recv:  8 | +0.008  -0.000  -0.009  -0.017  -0.025  -0.032  -0.042  -0.059  -0.070  -0.070
Recv:    |
Recv:  7 | +0.018  +0.009  +0.002  -0.012  -0.020  -0.033  -0.043  -0.049  -0.071  -0.071
Recv:    |
Recv:  6 | +0.052  +0.032  +0.011  +0.005  -0.004  -0.016  -0.023  -0.024  -0.038  -0.038
Recv:    |
Recv:  5 | +0.056  +0.033  +0.010  -0.010  -0.018  -0.017 [-0.015] -0.010  -0.016  -0.016
Recv:    |
Recv:  4 | +0.084  +0.056  +0.029  +0.003  -0.019  -0.020  -0.020  -0.012  -0.015  -0.015
Recv:    |
Recv:  3 | +0.075  +0.055  +0.035  +0.014  -0.003  -0.008  -0.015  -0.005  -0.008  -0.008
Recv:    |
Recv:  2 | +0.060  +0.045  +0.031  +0.022  +0.009  +0.004  -0.000  +0.000  -0.002  -0.002
Recv:    |
Recv:  1 | +0.069  +0.050  +0.032  +0.024  +0.009  +0.000  -0.010  -0.024  -0.037  -0.037
Recv:    |
Recv:  0 | +0.078  +0.055  +0.033  +0.026  +0.009  +0.000  -0.010  -0.024  -0.037  -0.037
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:     ( 10, 10)                                                              (220, 10)
Recv: 
Recv: Mesh is valid
Recv: Storage slot: 0
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height 10.00

Send: M500
Recv: echo:Settings Stored (682 bytes; crc 20880)
Recv: //action:notification Settings Stored
Recv: ok

Reboot

Send: M420 S1
Recv: echo:Invalid mesh.
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height 10.00

### Version of Marlin Firmware

 FIRMWARE_NAME:Marlin 2.0.9.3 (Jan 21 2022 18:54:19) 

### Printer model

Creality Ender 3 V2

### Electronics

Stock electronics

### Add-ons

BLTOUCH

### Bed Leveling

UBL Bilinear mesh

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7916545/Configuration.zip)

billygr, 2022-01-22:
Tried bugfix-2.0.x and getting Watchdog Reset

Recv: start
Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
Send: N0 M110 N0*125
Recv:  Watchdog Reset
Recv: Marlin bugfix-2.0.x
Recv: echo: Last Updated: 2022-01-21 | Author: Jacob Myers
Recv: echo: Compiled: Jan 22 2022
Recv: echo: Free Memory: 56327  PlannerBufferBytes: 1600
Recv: 
Recv: DWIN handshake error.
Recv: 
Recv: DWIN handshake //action:notification Ender-3 V2 Ready.
Recv: 
Recv: DWIN handshake echo:V86 stored settings retrieved (682 bytes; crc 20880)
Recv: Unified Bed Leveling System v1.01 active
Recv: Mesh loaded from slot 0
Recv: Mesh 0 loaded from storage.
Recv: //action:prompt_end
Recv: BL24CXX Check succeeded!

The-EG, 2022-01-22:
> G29 P1 ; automatically populate mesh with all reachable points
> G29 P3 ; infer the rest of the mesh values
> G29 P3 ; infer the rest of the mesh values again

> M420 S1 V ; enabled leveling and report the new mesh
> ...
> Recv: Mesh is valid
> Recv: Storage slot: 0
> ...
Send: M500
> ...
> Reboot
> ...
> Send: M420 S1
> Recv: echo:Invalid mesh.

You never saved the mesh to a slot. UBL doesn't save the mesh to EEPROM settings, so the only thing that is saved is the active slot. But since you didn't save anything to that slot, there was nothing to load once you rebooted and you got an invalid mesh instead.

Use `G29 Sn` and `G29 Ln` to save and load meshes, respectively. See https://marlinfw.org/docs/gcode/G029-ubl.html

billygr, 2022-01-22:
Will give a try as we speak, the Watchgod was solved by enabling #define EEPROM_INIT_NOW

ellensp, 2022-01-22:
check the number on your controller cpu.  creality has been swapping the cpu for one with half as much flash, causing many headaches
https://www.th3dstudio.com/2022/01/16/creality-4-2-x-cpus-512k-swapped-out-for-256k-warning-firmware/

billygr, 2022-01-22:
The printer is 1 year old, 4.2.2 Motherboard, hopefully this is not the case

billygr, 2022-01-22:
> 

You are right, with the G29 S G29 L it works after reboot. At least in bugfix-2.0.x.

I was expecting that the Send: M420 S1 V
....
Recv: Mesh is valid
Recv: Storage slot: 0
Followed by M500 will save the mesh

The-EG, 2022-01-22:
Yep, that's just how UBL works. ABL saved with M500, so perhaps you were using that before and got the two mixed up? I'd recommend sticking with UBL in any case, though.

And, FWIW once you've saved with `G29 S` and saved settings with `M500`, it *should* automatically load the mesh back in after reboot without requiring `G29 L`, but it never hurts to do it anyway.

billygr, 2022-01-22:
Just cleared it up, and proposed a change also to OctoPrint-BedLevelVisualizer examples (because this triggered everything).

Learned a couple things today . Thanks @The-EG

closed 2022-01-22 by billygr