MarlinFirmware/Marlin #23583: [BUG] M0 & M1 ignore string parameter

dmarnel, 2022-01-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The gcode commands M0 & M1 ignore the string parameter if given.

### Expected behavior

Issuing the gcode command M0 or M1 with a string message should display the message on the LCD screen.

### Actual behavior

A hard coded string message "M0 Stop" or "M1 Stop" is displayed.

### Steps to Reproduce

Edit any gcode file and insert an M0 command somewhere:

M0 Waiting...

When the command is processed it will display:

M0 Stop

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Creality Ender 3 Pro

### Electronics

BIGTREETECH_TFT35_V3.0

### Your Slicer

Cura

### Host Software

OctoPrint

ellensp, 2022-01-25:
This only works of 128x64 lcd's so only work when your lcd is in 128x64 mode.

It also works fine.

dmarnel, 2022-01-25:
@ellensp I don't know why you closed this. The parser.string_arg is not even being used to display the stop message, regardless of the LCD. To fix this, I changed line 88 in M0_M1.cpp from:

`TERN_(HOST_PROMPT_SUPPORT, hostui.prompt_do(PROMPT_USER_CONTINUE, parser.codenum ? F("M1 Stop") : F("M0 Stop"), FPSTR(CONTINUE_STR)));`

to:

` TERN_(HOST_PROMPT_SUPPORT, hostui.prompt_do(PROMPT_USER_CONTINUE, F(ui.status_message), FPSTR(CONTINUE_STR)));`

After making this change, the custom stop message is correctly displayed on my LCD, as well as in the popup message from Octoprint on my computer.

ellensp, 2022-01-26:
Because you never mentioned anything about  HOST_PROMPT_SUPPORT, and you didn't provide config files to try. I can only work with what I'm given.

You said "Issuing the gcode command M0 or M1 with a string message should display the message on the LCD screen."

I looked at the command with a lcd, and it is displayed on the lcd without issues.

Domush, 2022-02-08:
I made quite a few changes to the Host Actions/Prompt system, including fixing this Issue.

### Host action & Host prompt overhaul.
- Prompts are now informative.
- Actions which don't need user feedback are now notifications.
- Prompt timeouts now correctly close the prompt.
- M108 command now closes M0/1 prompts on host.
- Renamed host_actions arguments to be more intuitive

closed 2022-04-20 by github-actions[bot]