MarlinFirmware/Marlin #23586: [BUG] LCD_SERIAL_PORT not being properly set when using DWIN_CREALITY_LCD on BOARD_BTT_SKR_MINI_E3_V3_0

psuchris, 2022-01-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I installed the new BigTreeTech SKR mini E3 V3.0 (`STM32G0B1RET6`) onto my Ender 3 V2. This board was added in the 2.0.9.3 release. https://github.com/MarlinFirmware/Marlin/pull/23283

I modified the ribbon cable to be able to use the stock DWIN LCD, and also changed the relevant pins in `pins_BTT_SKR_MINI_E3_V3_0.h`.

I still was unable to get the LCD to display anything. I figured it had something to do with a `SERIAL_PORT` being set wrong, but it took me a while to find the actual cause was just that this board was never added to line 512 of `Conditionals_LCD.h`, which defines the `LCD_SERIAL_PORT`.

```cpp
// E3V2 extras
#if HAS_DWIN_E3V2 || IS_DWIN_MARLINUI
  #define SERIAL_CATCHALL 0
  #ifndef LCD_SERIAL_PORT
    #if MB(BTT_SKR_MINI_E3_V1_0, BTT_SKR_MINI_E3_V1_2, BTT_SKR_MINI_E3_V2_0, BTT_SKR_E3_TURBO) //HERE
      #define LCD_SERIAL_PORT 1
    #else
      #define LCD_SERIAL_PORT 3 // Creality 4.x board
    #endif
  #endif
  #define HAS_LCD_BRIGHTNESS 1
  #define LCD_BRIGHTNESS_MAX 250
#endif
```

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Conditionals_LCD.h#L512

So, in summary, line 512 of `Conditionals_LCD.h` is currently:
```cpp
    #if MB(BTT_SKR_MINI_E3_V1_0, BTT_SKR_MINI_E3_V1_2, BTT_SKR_MINI_E3_V2_0, BTT_SKR_E3_TURBO)
```
but should be:
```cpp
    #if MB(BTT_SKR_MINI_E3_V1_0, BTT_SKR_MINI_E3_V1_2, BTT_SKR_MINI_E3_V2_0, BTT_SKR_MINI_E3_V3_0, BTT_SKR_E3_TURBO)
```

To note, this is the same area of code relevant to this bug which I also ran into: https://github.com/MarlinFirmware/Marlin/issues/22299

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Creality Ender 3 V2

### Electronics

BigTreeTech SKR mini E3 V3.0

thisiskeithb, 2022-01-22:
Thanks for the heads up! I've submitted a PR: https://github.com/MarlinFirmware/Marlin/pull/23593

closed 2022-01-22 by thisiskeithb

bertos03, 2022-02-20:
For me, adding the "BTT_SKR_MINI_E3_V3_0" to line 512 did not do the trick: still exiting with a compiler error.
I commented out the "#define SERIAL_CATCHALL 0" by trial-and-error (after everything else had no success). That worked. Now, the FW compiles and my stock display is running.
Config is Ender 3 V2 together with SKR MINI E3 V3.0.

psuchris, 2022-02-27:
Hey @bertos03, the issue you're describing is actually a separate problem caused when defining multiple serial ports. I mentioned it at the very end of the original post: https://github.com/MarlinFirmware/Marlin/issues/22299

I ran into the same problem and happened to find the solution you stumbled upon in the comments of that other issue.