MarlinFirmware/Marlin #23591: [BUG] PROBE_OFFSET_WIZARD quit after homing because PAUSE_BEFORE_DEPLOY_STOW

brunosso, 2022-01-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying to accomplish the Z probe offset wizard, but when i start the wizard from LCD the first thing is an XYZ homing and because i have a Z probe with Optical Endstop (like this https://www.thingiverse.com/thing:3212906) i have PAUSE_BEFORE_DEPLOY_STOW activated.
So after the Z homing i click to stow the probe and the LCD is waiting in the Status screen. 
So the wizard is passed

I try to do before the homing XYZ and then the wizard, but the wizard ignore the actuale known/unknown position

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x (Jan 22 2022 10:59:57)

### Printer model

Creality Ender 3 Pro

### Electronics

MKS GEN L 2.1

### Bed Leveling

ABL Bilinear mesh

### Host Software

OctoPrint

### Additional information & file uploads

Configuration.h
https://pastebin.com/cyMH2BGx

Configuration_adv.h
https://pastebin.com/uXyBviMn

SidSkiba, 2022-01-22:
Wondering what type of LCD you are using?

lucaf86, 2024-05-03:
I've implemented a workaround to use the probe offset wizard with manual deploy/stow probes.
Even if it's not a complete implementation of the feature, because it requires a manual homing of all the axis before entering the wizard menù, it works.
Shortly, I inserted a check on all axis homed at the beginning of goto_probe_offset_wizard(), if homed continues otherwise exits without doing nothing and prints a message on the lcd. Then I removed the set_all_unhomed(); and the G28 command.
Doing this change and entering with all axis homed in the probe offset wizard now asks to deploy the probe and after confirmation enters the z hight adjustments screen.

```
void goto_probe_offset_wizard() {
#if ENABLED(PAUSE_BEFORE_DEPLOY_STOW)
  if (!all_axes_homed()) {
    FSTR_P const msg_fstr = GET_TEXT_F(MSG_HOME_AXES_FIRST);
    ui.return_to_status();       // To display the new status message
    ui.set_max_status(msg_fstr);
  }
  else {
#endif

    ui.defer_status_screen();
  #if DISABLED(PAUSE_BEFORE_DEPLOY_STOW)
    set_all_unhomed();
  #endif

    // Store probe.offset.z for Case: Cancel
    z_offset_backup = probe.offset.z;

    #ifdef PROBE_OFFSET_WIZARD_START_Z
      probe.offset.z = PROBE_OFFSET_WIZARD_START_Z;
    #endif

    // Store Bed-Leveling-State and disable
    #if HAS_LEVELING
      menu_leveling_was_active = planner.leveling_active;
      set_bed_leveling_enabled(false);
    #endif

  #if DISABLED(PAUSE_BEFORE_DEPLOY_STOW)
    // Home all axes
    queue.inject_P(G28_STR);
  #endif

    ui.goto_screen([]{
  #if DISABLED(PAUSE_BEFORE_DEPLOY_STOW)
      _lcd_draw_homing();
  #endif
      if (all_axes_homed()) {
        z_offset_ref = 0;             // Set Z Value for Wizard Position to 0
        ui.goto_screen(prepare_for_probe_offset_wizard);
        ui.defer_status_screen();
      }
    });

#if ENABLED(PAUSE_BEFORE_DEPLOY_STOW)
  }
#endif
}
```