MarlinFirmware/Marlin #23596: Creality – 4.2.X CPUs – 512K swapped out for 256K – WARNING POTENTIAL FIRE HAZARDS

garaysantiago, 2022-01-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

In Discord server a user was working on their machine and was having issues with the firmware. They posted a picture of the board and it revealed a lesser STM32 part number on that board. The 4.2.2 and 4.2.7 boards have all shipped with the STM32F103 RET6 CPUs since release. Now there is confirmation that 4.2.2 boards have shipped out from Creality with lesser 256K parts on them that are NOT safe to use with current Marlin and Unified firmware builds.

There are 4.2.2 (we would *assume* that 4.2.7s may have the lesser CPUs soon, we will be adding part number checks to the 4.2.7s that we sell going forward to make sure they have 512K parts on them) out in the wild with STM32F103 RCT6 CPUs. These CPUs are 256K parts whereas the RET6 versions were 512K parts. 

### Version of Marlin Firmware

2.0.9.3

### Printer model

Ender 3 Pro

### Electronics

4.2.2 and 4.2.7 boards have all shipped with the STM32F103 RET6 CPUs since release

thisiskeithb, 2022-01-22:
Have any other hardware changes been made? Adding a 256K variant is straight forward.

tome9111991, 2022-01-22:
https://www.th3dstudio.com/2022/01/16/creality-4-2-x-cpus-512k-swapped-out-for-256k-warning-firmware/

jbuck2005, 2022-01-22:
Pardon my ignorance but why the flag "potential fire hazard" ?
Are you proposing that the reduced memory space will force users to forgo the thermal runaway protection mechanism?

thisiskeithb, 2022-01-22:
> Pardon my ignorance but why the flag "potential fire hazard" ?

Overflashing could *potentially* lead to unexplained lockups with the heaters on.

I didn’t experience that issue personally, but some folks did, so YMMV.

jbuck2005, 2022-01-22:
Oh wow. Fair point!! I had not even considered that the firmware flash procedure would fail to check for out-of-bounds writes. Good lord.

thisiskeithb, 2022-01-22:
To reduce the size in your builds, change `genericSTM32F103RE` to `genericSTM32F103RC`: https://github.com/MarlinFirmware/Marlin/blob/02b01c773eb59b3495bf4e6ec7f71f1d4d97a4e1/ini/stm32f1.ini#L110

thisiskeithb, 2022-01-22:
I've created a PR, so please test & report any issues: https://github.com/MarlinFirmware/Marlin/pull/23599

closed 2022-01-22 by thisiskeithb