MarlinFirmware/Marlin #23598: [BUG?] Linear Advance Factor ignore Classic XYZ Jerk and max acceleration

PMGZED, 2022-01-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello 
I have Marlin FIRMWARE_NAME:Marlin 2.0.9.3 (Jan 21 2022 19:51:32)
With Classic Jerk #define CLASSIC_JERK
If i activate Linear Advance Factor with M900 K0.05 Marlin Ignore the Jerk of XYZ Axis and max acceleration
If i disable with M900 K0.00 all is fine.

If i have activate M900 K0.05 and set a E Jerk M205 B0.00 S0.00 T0.00 X0.00 Y0.00 Z0.00 E15.0 the E Jerk wil not ignored speed ofx xyz Axis accelration is ok.
if i set M205 B0.00 S0.00 T0.00 X10.00 Y10.00 Z0.00 E0.0 the Jerk and max acceleration parameter is ignored. The acceleration is fullspeed and i have jumping steps

### Steps to Reproduce

Compile Marlin with #define CLASSIC_JERK
Set M205 B0.00 S0.00 T0.00 X10.00 Y10.00 Z10.00 E0.00
M204 P500.00 R3000.00 T3000.00

Activate Linear Advance Factor 
M900 K0.05 

### Version of Marlin Firmware

2.0.9.3

### Printer model

Modify German RepRap X400 with GTR V1.0

### Electronics

GTR V1.0 TMC2130

### Your Slicer

Simplify3D

### Host Software

Repetier Host

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7919122/Configuration.zip)

thinkyhead, 2022-04-21:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

tombrazier, 2023-08-20:
There are several things to say about this.

First, jerk is only used to calculate cornering speed whether you are using classic jerk or junction deviation. Accelerating from a stop does not jump to the jerk speed. With classic jerk, this is a bug I think. Certainly the classic jerk code calculates a start speed when there are no previous segments executing but it is ignored and `MINIMUM_PLANNER_SPEED` is used instead. I am working on a PR that fixes this and also sets a more "correct" minimum speed than `MINIMUM_PLANNER_SPEED`.

Second, acceleration for X, Y and Z is capped for print movements when linear advance is active. This is necessary because the _speed_ at which filament is extruded for linear advance is proportional to the _acceleration_ of the movement. A sudden in X, Y, Z acceleration must be matched by a sudden jump in E speed. And jumps in E speed are governed by the E jerk parameter.

I think this is the behaviour that is being described in the OP. This is not a bug. It is by design. However there is a better way. Instead of using E jerk, it would be possible (but I think computationally more expensive) to accelerate E at its maximum rate and only cap the X, Y and Z acceleration for a short time at the start of a segment. I would like to think I'll be able to submit a PR for this at some point.

Incidentally, while I am at it, it may be worth mentioning that when junction deviation is being used, it is also used to calculate E jerk in a way that does not make sense. Just in case this confuses anyone else. I don't recommend junction deviation, though, because jerk is the physically correct way to calculate cornering speed. For certain corners JD can result in very high jerks, far more than the machine can handle.