MarlinFirmware/Marlin #23600: [BUG] (MKS Robin Lite - Bootloop when SD-Card Inserted)

ETE-Design, 2022-01-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to update an MKS Robin Lite board to newest Marlin Bugfix 2.0.x, using standard example file with changes needed to fit an MKS Robin Lite board... Compiling + Uploading is fine... Problem is when you insert the SD-Card the Board reboot after about 5-7 second, that keeps happen until the SD-Card is pulled out again.

### Expected behavior

When SD-Card Inserted nothing should happend.

### Actual behavior

Reboot after 5-7 Second after SD-Card in inserted and do it until the SD-Card is removed again.

### Steps to Reproduce

1. Update to Bugfix 2.0.x / Marlin 2.0.9.3
2. Insert SD-Card
3. Wait for Reboot / Bootloop

### Version of Marlin Firmware

Bugfix 2.0.x

### Electronics

MKS Robin Lite

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7919426/Configuration.zip)

sleepsemek, 2022-01-25:
I have the same problem with BTT SKR MINI V1.1. Sd card in both mini12864 LCD sd slot and motherboard sd slot causes bootloop

ETE-Design, 2022-01-30:
@sleepsemek Did you manage to solve the problem?

sleepsemek, 2022-01-30:
> @sleepsemek Did you manage to solve the problem?
Try to add in SD card section of your pins_"boardname".h 
#define ENABLE_SPI2

ETE-Design, 2022-01-31:
@sleepsemek Seems like not working here :-(

closed 2022-02-02 by ETE-Design