MarlinFirmware/Marlin #23602: [FR] Display for 4 axis foam cutter

rcKeith, 2022-01-23:
### Is your feature request related to a problem? Please describe.

I'm using Marlin 2.0.9.3 to drive a 4 independent axis foam cutter. I seem to have the code working as I want, I've not tested it on the machine yet just have it running on the Mega 2560 and RAMPS1.4 board.  I'd like to be able to display the 4th axis in the Status screen with XY and Z.  My 4th Axis is U but internally reference as AXIS4.  I've looked at the code in \Marlin\src\lcd\dogm\dogm_Statusscreen.h but not sure how I can add the 4th axis.
I've watched  Thinkyhead video's which are very good and helped a bit but I am a bit stumped.

Any help is greatly appreciated 
Keith

### Describe the feature you want

4th Axis displayed in Status screen

DerAndere1, 2022-01-30:
The questioin is in what position on the screen do you want to get the axes displayed. With the current layout there is no space for additional axes. What do you want to replace, the icons for hotends, bed and fans etc.? or the print progress?

rcKeith, 2022-01-30:
The Fan is not needed for a foam cutter so that would be a suitable place.  Thanks for the reply.
Keith

DerAndere1, 2022-01-31:
Are you aware of the `FOAMCUTTER_XYUV` kinematics? it is a not so well documented feature of Marlin. if you `#define FOAMCUTTER_XYUV` and you have condifgured 5 linear axes, you can configure the correct axis names (X,Y,U and V) and Marlin takes care of proper feedrate interpretation with such a machine. Also, if you use a fan pin to control the heat of the tool, you can change the heat with M106 Sxxx where xxx is the percentage of maximum heat. User hobbyseven obtained good results when he used the JediCut CAM software to create toolpaths for a machine with `FOAMCUTTER_XYUV` enabled.

Regarding the display of additional axes: I have no experience with the display code and no means to test it. But I drafted an experimental implementation where I expect that all icons at the top of the screen are replaced by the positions of the extra axes. Please test this branch: https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot_edge . the relevant commit is https://github.com/DerAndere1/Marlin/commit/b91e9570a0f46a237e4209500c0b2f9aab334ae9 . In your configs, you will have to do the following:
- Add `#define FOAMCUTTER_XYUV`
- rename `LINEAR_AXES` to `NUM_AXES`
- `#define NUM_AXES 5`
- `define AXIS4_NAME 'U'`
- `define AXIS5_NAME 'V'`
- make sure you **don't** have the new option AXIS4_ROTATES or AXIS5_ROTATES defined
- the Z axis is unused. A dummy value can be assigned to Z axis pins
- enable the new option `#define LCD_SHOW_SECONDARY_AXES` in Configuration_adv.h

rcKeith, 2022-01-31:
Many thanks, I will give it a try later today and let you know.
Keith