MarlinFirmware/Marlin #23605: [BUG] (bug summary)Power Loss Recovery unable to enable(Ender3 v2)

wanyulaowang, 2022-01-23:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

After I transferred from the official firmware to Marlin 2 0.9.3 firmware, the printer's power loss recovery cannot be enabled, and there is still a failure when trying to recompile
I learned from Google that the first few layers may not be able to use this function, so I tried to wait and manually power off and restart at more than 3000 layers, but there was still no prompt
I checked the configuration_ Configuration of adv.h file, "#define power_loss_recovery" is enabled

### Bug Timeline

reload the new firmware

### Expected behavior

After manual power off and restart, the power loss recovery prompt appears

### Actual behavior

No prompt after restart

### Steps to Reproduce

1.Start printing
2.Turn off the power manually
3.Re energize
4.No hint

### Version of Marlin Firmware

2.0.9.3

### Printer model

ender3 v2 

### Electronics

Stock electronics(Motherboard model V422)

### Your Slicer

Prusa Slicer

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7921023/Marlin.zip)

tome9111991, 2022-01-23:
You can try 

Change
` #define PLR_ENABLED_DEFAULT   false`
To
` #define PLR_ENABLED_DEFAULT   true`

In
configuration_adv.h

josederas, 2022-01-24:
For reviews clarification, did you enable Power Loss Recovery on the UI?  The option is found under the Control menu item. The default setting is OFF.

closed 2022-01-26 by wanyulaowang