MarlinFirmware/Marlin #23608: A lot of bugs on MKS UI in the Bugfix branch

DvoraNoob, 2022-01-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After the PR: 23464 merged my encoder doesn't work more, and the screen stays yellow

The TFT_LVGL_UI Z Offset Wizard have 2 problems. First one, compile error  because the name of variable. Second after I click on Wizard Menu and in sequence I click on Leveling Offset Probe Menu my board reset.

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x (Jan 23 2022 18:51:17) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Electronics

Robin Nano V3

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

descipher, 2022-01-24:
try disabling SPI_DMA_TC

pins_MKS_ROBIN_NANO_V3_common.h

  372,11:   #define USE_SPI_DMA_TC <-Comment this line

DvoraNoob, 2022-01-24:
> try disabling SPI_DMA_TC
> 
> pins_MKS_ROBIN_NANO_V3_common.h
> 
> 372 #define USE_SPI_DMA_TC <-Comment this line

I tried it, I got a compilation error

descipher, 2022-01-24:
Not good. That should not happen based on the pins file. 
What line number(s) failed.

DvoraNoob, 2022-01-24:
> ‎Não é bom. Isso não deve acontecer com base no arquivo pinos. ‎‎ Qual número de linha(s) falhou.‎

Now compiled, but I'll try before confirm.

DvoraNoob, 2022-01-25:
> Not good. That should not happen based on the pins file. What line number(s) failed.

I got a bootloop.

descipher, 2022-01-25:
Definitely broken then, you should be able to turn it off.

DvoraNoob, 2022-01-25:
> ‎Definitivamente quebrado, então, você deve ser capaz de desligá-lo.‎

I went back to Marlin Branch

descipher, 2022-01-25:
Confirmed bug, just repeated bootloop on bugfix_2.0.x with configuration.h as:
LCD is MKS TS35 V2.0

#define BOARD_MKS_ROBIN_NANO_V2
#define SERIAL_PORT 3
#define SERIAL_PORT_2 1
#define TFT_GENERIC
#define TFT_INTERFACE_SPI
#define TFT_RES_480x320
#define TFT_LVGL_UI
#define SDSUPPORT

DvoraNoob, 2022-01-25:
> 

I'll need change to V3 because my board is a v3, but I have v2 too, but without use

Can I use USB?

descipher, 2022-01-25:
The boot loop on V2 demonstrates that the SPI code is broken since V2 do not have USE_SPI_DMA_TC defined. 
Question: Did you update the assets?

DvoraNoob, 2022-01-25:
> 
With TFT-Generic settings I got a combilation error

Yes, I update the assets everytime.

descipher, 2022-01-25:
Please upload you config files.

descipher, 2022-01-25:
TST35

DvoraNoob, 2022-01-25:
https://1drv.ms/u/s!AlJymIpivUgt0CF9jtnn1LigVlne?e=xLm8A1

DvoraNoob, 2022-01-28:
FIXED

closed 2022-01-28 by DvoraNoob