MarlinFirmware/Marlin #23609: [BUG] TFT LCD prevents building

BlockArchitech, 2022-01-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I enable my TFT let, color ui, and touchscreen support I get:

```
In file included from Marlin/src/HAL/STM32/tft/tft_fsmc.cpp:28:
Marlin/src/HAL/STM32/tft/tft_fsmc.h:55:12: error: 'SRAM_HandleTypeDef' does not name a type; did you mean 'UART_HandleTypeDef'?
   55 |     static SRAM_HandleTypeDef SRAMx;
      |            ^~~~~~~~~~~~~~~~~~
      |            UART_HandleTypeDef
In file included from Marlin/src/HAL/STM32/tft/tft_fsmc.cpp:28:
Marlin/src/HAL/STM32/tft/tft_fsmc.h:100:12: error: 'FSMC_NORSRAM_DEVICE' was not declared in this scope
  100 |   {PD_14,  FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D00
      |            ^~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/STM32/tft/tft_fsmc.h:101:12: error: 'FSMC_NORSRAM_DEVICE' was not declared in this scope
  101 |   {PD_15,  FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D01
      |            ^~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/STM32/tft/tft_fsmc.h:102:12: error: 'FSMC_NORSRAM_DEVICE' was not declared in this scope
  102 |   {PD_0,   FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D02
      |            ^~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/STM32/tft/tft_fsmc.h:103:12: error: 'FSMC_NORSRAM_DEVICE' was not declared in this scope
  103 |   {PD_1,   FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D03
  ```

And the build fails.

I have zero idea why this is.

### Bug Timeline

New bug, today

### Expected behavior

The build to finish

### Actual behavior

The build fails.

### Steps to Reproduce

1: Enable my current TFT LCD
2: Enable Color UI
3: Enable Touch Support
4: Build Marlin

### Version of Marlin Firmware

2.0.x

### Printer model

QIDI x-one2

### Electronics

Stock

### Add-ons

None

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/7922080/Archive.zip)

ellensp, 2022-01-24:
I notice configs are 02000901

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

ellensp, 2022-01-24:
You have selected BOARD_CHITU3D with a TFT_TRONXY_X5SA
But the motherboard BOARD_CHITU3D does not seem to support that type of screen
Are you sure you dont have a BOARD_CHITU3D_V5 or BOARD_CHITU3D_V9 as these do support it.

BlockArchitech, 2022-01-24:
This worked! Thanks. Could we add some sort of `Did you mean to use BOARD_CHITU3D_V5?`?

ellensp, 2022-01-24:
The motherboard is the most important thing you must get right.

closed 2022-01-24 by ellensp