MarlinFirmware/Marlin #23611: [BUG] ASSISTED_TRAMMING

Snargola, 2022-01-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm attempting to set up and use Assisted Tramming. Neither the comments in the code nor the Marlin documentation cover this feature very well. It is not explained if the probe points need to have the nozzle or the probe over the bed screws. If it is the probe, then this is not possible on the right side as there is not enough room to move the X gantry in that direction. If I use the nozzle, compiling the code throws an error stating the following:

Marlin\src\gcode\bedlevel\../../feature/tramming.h:36:71: error: static assertion failed: TRAMMING_POINT_XY point 2 is not reachable with the default NOZZLE_TO_PROBE offset and PROBING_MARGIN.

If I use the coordinates included in the code it works, but, neither the nozzle nor the probe are over the bed screws.

Long story short: This is probably not a bug, I just don't know the proper coordinates to put in.

This is on a Creality Ender 3. Bed screws are located at:
Front Left: 30x 30y
Front Right: 200x 30y
Back Right: 200x 200y
Back Left: 30x 200y

My BLTouch is offset -47

Thank you in advance for any and all help provided.

### Bug Timeline

New. It started today.

### Expected behavior

I would expect the machine to be able to assist me in tramming the corners.

### Actual behavior

If it actually probes, neither the nozzle nor the probe are over the bed screws. If I use coordinates putting the nozzle over the screws, the compiler complains about that.

### Steps to Reproduce

1) Try to use coordinates that make sense.
2) Watch compiler error.

### Version of Marlin Firmware

bugfix 2022-01-22

### Printer model

Creality Ender 3

### Electronics

BTT SKR Mini E3 v2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

ellensp, 2022-01-25:
"My BLTouch is offset -47"
In X, or Y or both? Presuming in X. wich means the mac position you can probe is your X bed zise -47

ellensp, 2022-01-25:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-01-25 by ellensp

sarvenn, 2022-02-21:
@Snargola You may change your X and Y max position to your mechanical limits to probe as much as areas you could.
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE

For example, I use an Ender 3 pro and my X max is set to 249, Y max is set to 244. Y home offset is set to -4 (at home position nozzle is 4 mm away from the bed edge, therefore it is set to -4mm (not in configuration.h, I send gcode command after updating my fw). 

First of all you need to set your phsyical boundaries and then set your bed size.