MarlinFirmware/Marlin #23617: [BUG] (SD problem on skr pro V1.2)

Enzo-Enz, 2022-01-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi
Marlin 2.0.9.3 doesn't show the file on the SD card
I use BTT SKR PRO V1.2 as motherboard and generic delta configuration for delta printer
I have enabled enabled SDSUPPORT and SD_CHECK_AND_RETRY but when I click on "print from media" It doesn't show the file on SD card.I test bugfix version but the issue still exist
there are g-codes on SD card

here are the pictures
[image]

[image]
[image]

### Bug Timeline

new

### Version of Marlin Firmware

2.0.9.3

### Printer model

DIY delta printer

### Electronics

skr pro v1.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

borland1, 2022-01-26:
Not sure which LCD you have.  Maybe your LCD display setup is for REPRAP_DISCOUNT_SMART_CONTROLLER which has SD-card reader. 

You can enable the onboard SD-card slot in Configuration_adv.h with this define...
  
`  #define SDCARD_CONNECTION ONBOARD`

```
       /**
       * Set this option to one of the following (or the board's defaults apply):
       *
       *           LCD - Use the SD drive in the external LCD controller.
       *       ONBOARD - Use the SD drive on the control board.
       *  CUSTOM_CABLE - Use a custom cable to access the SD (as defined in a pins file).
       *
       * :[ 'LCD', 'ONBOARD', 'CUSTOM_CABLE' ]
       */
1629   //#define SDCARD_CONNECTION LCD
```

thisiskeithb, 2022-01-26:
> I use BTT SKR PRO V1.2 as motherboard and generic delta configuration for delta printer
I have enabled enabled SDSUPPORT and SD_CHECK_AND_RETRY

Are there any other changes? Please attach the modified configs you used to test the latest bugfix.

thisiskeithb, 2022-01-27:
> I enabled `#define SDCARD_CONNECTION LCD` the issue fixed

[I explained this already in your last issue](https://github.com/MarlinFirmware/Configurations/issues/652#issuecomment-1008282148). This is not a bug.

> but for my XYZ printer(ramps+mega2560+12864 lcd with SD-card reader+marlin 2.0.9.3) I just enabled `SDSUPPORT` and `SD_CHECK_AND_RETRY` and there was no issue!

Awesome!

> also there was no issue with previous version(2.0.9.2) with this settings(`SDSUPPORT` and `SD_CHECK_AND_RETRY`)for my delta

Great!

closed 2022-01-27 by thisiskeithb