MarlinFirmware/Marlin #23618: [BUG] BTT SKR2 F_429 uncontrolled RESET

qulet, 2022-01-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have problems with my BIGTREETECH SKR 2 board with a STM32 F429 chip on it.
I am using the current "Marlin-bugfix-2.0.9.2.x" branch from just now.
Performs builds for the BIGTREE_SKR_2_F429 environment.
Unfortunately, Marlin with a connected printer on TCM2208 seriall controllers, after executing "Home", there is an uncontrolled RESET
This is my second board advertised with the F429 processor. ...... The complaint is an uncontrolled resetting of Marlin.
I have replaced the LCD smart controller.
No stepsticks
the board also resets. Reading the temperature, heating, engine operation, homing, everything works for a while until reset.

I don't know where the problem is ...
hardware or software

link to the video:

https://drive.google.com/file/d/1cfCH6-XSG8p-vmQ455tRZ7oQbPEW9kO-/view?usp=sharing

### Bug Timeline

uruchomienie pierwszego wydruku

### Expected behavior

work without reset

### Actual behavior

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/7936124/Config.zip)
https://drive.google.com/file/d/1cfCH6-XSG8p-vmQ455tRZ7oQbPEW9kO-/view?usp=sharing

### Version of Marlin Firmware

Marlin-bugfix-2.0.9.2.x

### Printer model

DIY Core XY

### Add-ons

Sonda Z servo, aktywna opcja #define LASER_FEATURE

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Additional information & file uploads

Connecting...
Printer is now online.
>>> M501
SENDING:M501
echo:V85 stored settings retrieved (621 bytes; crc 42406)
echo:; Linear Units:
echo:  G21 ;
 (mm)
echo:; Temperature Units:
echo:  M149 C ; Units in Celsius
echo:; Filament settings (Disabled):
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo:  M92 X101.61 Y101.61 Z400.00 E393.15
echo:; Max feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:; Max Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S0 Z10.00 ; Leveling OFF
echo:; Material heatup parameters:
echo:  M145 S0 H210.00 B60.00 F0
echo:  M145 S1 H240.00 B110.00 F0
echo:; Hotend PID:
echo:  M301 P9.76 I0.41 D58.54
echo:; Controller Fan:
echo:  M710 S255 I0 A1 D60 ; (100% 0%)
echo:; Power-loss recovery:
echo:  M413 S0 ; OFF
echo:; Z-Probe Offset:
echo:  M851 X50.00 Y15.00 Z-0.70 ; (mm)
echo:; Stepper driver current:
echo:  M906 X550 Y550 Z600
echo:  M906 T0 E400
echo:; Hybrid Threshold:
echo:  M913 X100 Y100 Z15
echo:  M913 T0 E30
echo:; Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 T0 E
echo:; Filament load/unload:
echo:  M603 L0.00 U50.00 ; (mm)
echo:; Filament runout sensor:
echo:  M412 S0 ; Sensor OFF

oliver-eifler, 2022-01-25:
check your endstop wiring

ellensp, 2022-01-26:
Please do some basic diagnostics.
Ie connect up the serial port and check what it says there.

qulet, 2022-01-26:
Podałem odczyt z M501 z Programu Printrun  wszystko działa prawidłowo do momentu resetu. M119 OK . Drukarka wykonuje procedurę Bazowania "HOME" i chwile póżniej następuje Reset  . Wszystko w czsie żeczywistym jak na podlinkowanym filmie.

qulet, 2022-01-26:
The following were listed: REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
, 12V power supply. all of this did not help to correct the problem.
BTT SKR2 without stepsticks also reset.

qulet, 2022-01-28:
Found the reason for the Marlin reset.
Even though there is no compile error, "Zservo probe" "Laser control" cannot be run at the same time, this is the reason for the reset.
Compiling "Fix-Mounted Probe" and "Laser control" does not reset Marlin.
Can someone verify this?
The same problem occurs with MKS SBASE v1.3, but there pops up a message that this configuration cannot be compiled.

qulet, 2022-01-28:
Running the #define LASER FEATURE function causes the reset BTT SKR2 VB F-429

qulet, 2022-01-29:
Can you change something in the configuration files to make the problem disappear?

descipher, 2022-01-29:
Not sure what's causing the reset in the old code, I have not tested on an F429 but there is this PR you can try: https://github.com/descipher/Marlin/archive/refs/heads/laser.gcode.inline.fix.zip

qulet, 2022-02-02:
descipher
I compiled Marlin from the given link. Merlin resets after a few minutes. The user Dennis02121978 suggested how to avoid the reset. You need to remove the MicroSD card from the slot on the SKR2 board after uploading the firmware .. I confirm, I tested only on the plugged in TMC and the LCD display with an active laser function for 30 minutes there was no reset :)))

mikeww1, 2022-02-07:
Maybe I can add something to this issue. I have no laser but was exeperiencing regular reboots with the SKR2 board too.
What I did was disabling these parameters.
Configuration_adv.h
#define USB_FLASH_DRIVE_SUPPORT
#define USE_OTG_USB_HOST
#define NO_SD_DETECT
#define MULTI_VOLUME
No more reboots. I did not look any further into this issue as I dont need the flash drive support.

qulet, 2022-02-11:
mikeww1 
Marlin-laser.gcode.inline.fix version, I had these functions turned off, but when the SD card was in the SKR2 slot, it was reset. When the card was withdrawn it was about 1.5 hours, everything was OK! I tested without the extruder, temperature sensors and limit switches connected. 
Thanks for the info.

kursatu, 2022-03-01:
Only removing the sd card worked for me. 
I've uncommented  these first but it still reset itself.
#define USB_FLASH_DRIVE_SUPPORT
#define USE_OTG_USB_HOST

grawi-sikorsky, 2022-03-02:
Same here - random reset on F429 SKR2 version. 
All parameters mentioned here are disabled.
But in my case, it doesn't matter if i have microSD pluged or no - it keeps reset anyway randomly (after 3,15,30 minutes). I also tried to remove red jumpers from board used to switch between wifi and usb - no success.
On F407 version it works ok.

MiguelAReis, 2022-03-16:
I was also getting random reset, only when I disabled the SD Card the printer stopped resetting itself.

dalogan82, 2022-04-09:
HiI have the same problem, when the  board is working in 5 minuts shutdown!!

Randwic, 2022-04-28:
Hello,

I have the same problem.  when the sd card is plug into the motherboard, the printer restarts every 180 seconds.

ellensp, 2022-04-28:
All those saying "I have the same problem."  please include your config files, so we can  find commonalities etc

Randwic, 2022-04-28:
> All those saying "I have the same problem."  please include your config files, so we can  find commonalities etc  

Hello, config.h and config.adv.h?

ellensp, 2022-04-28:
Yes and any other files you changed

Randwic, 2022-04-28:
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/8583932/Config.zip)

qulet, 2022-04-30:
My configuration
[CONFIGURACJA.zip](https://github.com/MarlinFirmware/Marlin/files/8597027/CONFIGURACJA.zip)
:

davbcr, 2022-05-03:
> MULTI_VOLUME

I tried this and i doesn't work untill i stepped out the SD card but now i cannot get the SD flash procedure to start again with none of the tons of the SDs i have.
All of them are formatted to FAT32...... 
Do i need to flash the SKR2 via USB cable?

mojamuto, 2022-06-26:
I have the same problem, with the sd inserted restarts after 180 seconds.

[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/8986935/Configuration.h.zip)
[Configuration_adv.h.zip](https://github.com/MarlinFirmware/Marlin/files/8986936/Configuration_adv.h.zip)
\

qulet, 2022-08-28:
Karta się resetuje Marlin v. 2.09.05

closed 2022-10-30 by qulet