MarlinFirmware/Marlin #23621: [FR] (feature summary)Ender3 V2 new screen support( replaced DWIN with DACAI)

Li-Haung, 2022-01-26:
### Is your feature request related to a problem? Please describe.

I got a new ender3 V2 this year,but Creality replaced DWIN screen with DACAI.
Can add support for this screen ?

### Describe the feature you want

Add support for DACAI screen 

### Additional context

[image]

ellensp, 2022-01-26:
Some work has already been done on this see https://github.com/MarlinFirmware/Configurations/issues/645#
This screen is also supplied on Ender 3 S1

mriscoc, 2022-01-29:
This display emulates some of the functionalities of the old DWIN display but its API is not exactly the same. I found the programming interface for the old DWIN, but not for this new unit. Have you found some information about it?

ellensp, 2022-02-28:
the PR to support these ass been added. https://github.com/MarlinFirmware/Marlin/pull/23822
closing.

closed 2022-02-28 by ellensp