MarlinFirmware/Marlin #23623: [BUG] (bug summary)The printer stops when entering the main menu

soroush8266, 2022-01-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have a problem with lcd mks mini 12864 v3 When I click on the lcd button to enter the menu while printing, the printer stops for a few hundredths of a second and starts working again.
It's a bit slow to bring up the main menu, but it has no problem with the same lcd and normal mks mini1286 settings

### Bug Timeline

.

### Expected behavior

lcd work without stopping or taking time

### Actual behavior

lcd enter the main menu without stopping or taking time

### Steps to Reproduce

1.confing marlin
2.enable mks mini12864 v3
3.enable tmc2208

### Version of Marlin Firmware

marli. 2.0.9.3

### Printer model

Kossel

### Electronics

Mks sgenl v1+tmc2208uart

### Add-ons

.

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Additional information & file uploads

Zz.zip

soroush8266, 2022-01-26:
My friends, I checked if the define SDCARD CONNECTION LCD is disabled, the problem will be solved, and if it is active, the same problem still exists

soroush8266, 2022-01-27:
Please help me

ellensp, 2022-01-28:
closing as user has created new issue for the same thing, but slightly better worded.

closed 2022-01-28 by ellensp