MarlinFirmware/Marlin #23625: [BUG] inconsistant extrusion with arcs (G2/G3)

rerickson1, 2022-01-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Starting with release 2.0.9.2 when printing with arcs, extrusion is inconsistant.
At the end of the seam, it will under extrude or over extrude.
This issue didn't happen in 2.0.9.1. 

### Bug Timeline

2.0.9.2

### Expected behavior

print quality with arcs should match or exceed print quality without arcs.

### Actual behavior

print quality with arcs is sub-par.

### Steps to Reproduce

Slice with Prusa slicer and run through ArcWelder Lib.

### Version of Marlin Firmware

2.0.9.3

### Printer model

Anet A8 Plus with modifications

### Electronics

MKS SGen L V2

### Add-ons

Bondtech extruder with E3D V6 hotend

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

The firmware I am running is 2.0.9.3 with these changes: https://github.com/MarlinFirmware/Marlin/compare/2.0.9.3...rerickson1:rde/2.0.9.3_a8+_custom

Issue never happened with 2.0.9.1. Could this change have introduced the issue?
https://github.com/MarlinFirmware/Marlin/pull/22599

Pictures of the issues:

over and under extrusion at end of seam:
[image]

over extrusion at end of seam:
[image]

same print printed **without** arcs:
[image]

thinkyhead, 2022-01-26:
> Could this change have introduced the issue?

Try reverting that change to see if it helps.

rerickson1, 2022-01-28:
> Try reverting that change to see if it helps.

Reverting the change didn't help. There is still an issue.

What's very weird about this issue is that it only manifests after the firmware has run for a while.
See this print. The print takes about 3.5 hours at 0.2 mm layer height. The issue manifested towards the end of the print.
[image]

After this print was done, I re-printed the exact gcode and the issue showed throughout the print:
[image]

I reverted the commit and updated my printer with the new firmware. Of course this reset the printer so I was interested to see what would happen. The issue is much less prevalent, but is showing quality issues at the end of the seam.
[image]

Then right after that print was done, I started it again to see what the result would be after the printer has been running a while and the quality was worse.
[image]

As a reminder, If I do not print with arcs, these prints come out flawless.

rerickson1, 2022-02-21:
This issue does exist in 2.0.9.1 as well. Can anyone else reproduce this issue?

DiskDoctor, 2022-04-16:
Do you get this on a different printer? You may be experiencing HEAT CREEP

rerickson1, 2022-04-16:
> Do you get this on a different printer? You may be experiencing HEAT CREEP

If I print without arcs, this never happens. So I have to believe there are no hardware issues with the printer.

thinkyhead, 2022-04-20:
It would be interesting to see with isolated `G2`/`G3` moves whether the extruder is moving the same amount of filament as it does when doing the same with `G1`. Be sure to turn off any "advance extrusion" in your slicer when testing any of this stuff. Some slicers try to account for filament spring and back-pressure, but they are not able to do this well in relation to arcs.

rerickson1, 2022-04-21:
I believe I have found the source of my error buried in the slicer settings. I used PrusaSlicer 2.4.0 for the tests. For the initial testing I had `use_relative_e_distances = 1`.
I re-ran the tests now with PrusaSlicer 2.4.1 and set `use_relative_e_distances = 0`

closed 2022-04-21 by rerickson1