MarlinFirmware/Marlin #23626: Can we add an option for HR4988 drivers on the v4.2.2 board?

r0dnay, 2022-01-26:
### Description

Can we add an option for HR4988 drivers on the v4.2.2 board?

all the other driver options seem to be present including a workaround for the TMC2225 and TMC2226 drivers where the #define of another driver is advised

### Configuration Details
* Manufacturer: Creality
* Model: Ender 3 PRO
* Motherboard: V4.2.2

#### Additional Information
letter C on SD Card cover (HR4988)
    C = HR4988
    E = A4988
    A = TMC2208
    B = TMC2209
    H = TMC2225
* Provide links to the requested printer/configuration.
* https://www.th3dstudio.com/hc/product-information/3rd-party-control-boards/creality-boards/creality-v4-2-x-board-driver-codes/

ManuelMcLure, 2022-01-26:
HR4988 are not exactly the same as A4988. For one thing HR4988 supports up to 16x/32x/64x/128x micro stepping and are really designed to run at 32x or higher. 16x on an HR4988 is much noisier than 16x on an A4988. I had the misfortune of having HR4988s shipped to me instead of the A4988 modules I ordered early in my 3D printing career - the noise was unbearable in comparison to the A4988s I replaced them with.
However, per the datasheets of both drivers pulse timings are exactly the same between the A4988 and the HR4988.

r0dnay, 2022-01-27:
well that was my point, theyre not the same i know because of the way they behave in certain configurations so not really redundant as mentioned - was kinda hoping for someone, pref a dev/coder with an open mind and a logical thought process to chime in, i thought ellensp was the only one replying to all my posts :p

thanks thinkyhead, ive been in touch with creality with my findings and your extract and asked them for details - ill report back with their response if i get one, in the meantime ill try and find some more data on both stepper drivers

closed 2022-04-08 by github-actions[bot]

r0dnay, 2022-05-06:
ive not had that much luck getting creality to respond via the creality.com website - does anyone have a contact at creality? TeachingTech on Youtube said in his video which outlines how to identify boards, that he had been in touch with his contact at creality - but when i emailed him to see if he could help get some info, he told me he didnt have any contacts at creality :s 

i know that between marlin and creality something is different in the firmware between A4988 and HR4988, its audible and visible, either something is coded incorrectly or both HR and A variants are very different indeed

so if anyone has any contacts at creality, if you could drop a contact here or DM me and we can forward the email over

in the meantime ill try other creality sites, maybe try via the official creality amazon store

ManuelMcLure, 2022-05-06:
HR4988 are noisier than A4988 if you use them at 16x micro stepping because they are designed for 32x micro stepping. I ran into this when I replaced my Anet A8 original board (A4988) with RAMPS - although the seller claimed A4988 drivers they actually sent me HR4988 and the noise level was awful. When I actually got real A4988 the noise went down to the same level as my original Anet board.
Unless you can change the micro step setting for the drivers, I think you're out of luck.
[https://www.reddit.com/r/3Dprinting/comments/75w22r/psa_hr4988_drivers_cause_noise_and_vibration/](https://www.reddit.com/r/3Dprinting/comments/75w22r/psa_hr4988_drivers_cause_noise_and_vibration/)

mr-gosh, 2022-06-21:
I ran into that problem too and meanwhile there is no option in marlin to use a dedicated HR4988 driver - so we are stuck at proposing it is an A4988 only right? Or is there a way to configure the HR4988 more sane?

ManuelMcLure, 2022-06-21:
The thing is that as far as Marlin is concerned the only applicable settings for A4988 and HR4988 are the pulse timings, and those are identical. So there's no such thing a "more sane" option to use for an HR4988 on the Creality boards. The micro stepping is not configurable on those boards so there's nothing Marlin can do to change it.

mr-gosh, 2022-07-03:
> The micro stepping is not configurable on those boards so there's nothing Marlin can do to change it.

OK thanks for the reply - probably we should add that to the documentation of the compile how to - cause at the moment it reads like there should be an option selected that is not available if you have an HR4988.