MarlinFirmware/Marlin #23632: [BUG] (G33 delta auto calibration stuck on AC checking)

Voxel3DPrinting, 2022-01-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

hey, 
I have a custom delta printer and am running into issues with G33 auto calibrate..

It starts, says AC checking, probes 4 points and then raises and does nothing, on the LCD it keeps saying "AC checking"on pronterface it says ;

22:27:58.623 : G33 Auto Calibrate
22:27:58.625 : Checking... AC
22:27:58.628 : .Height:280.00  Ex:+0.00  Ey:+0.00  Ez:+0.00  Radius:125.63
22:27:58.630 : .               Tx:+0.00  Ty:+0.00  Tz:+0.00
22:28:03.622 : echo:busy: processing
22:28:08.622 : echo:busy: processing
22:28:13.657 : echo:busy: processing
22:28:18.656 : echo:busy: processing
22:28:21.834 : Correct delta settings with M665 and M666"

After that it just does nothing. What could the problem be? I have a fixed mounted probe, which works perfectly in G29....
I know for sure that my delta radius and height are pretty well dialed in...
anyone know why this happens and how to fix it?

### Expected behavior

> G33 Auto Calibrate
> Checking... AC
> .Height:295.00    Ex:+0.00  Ey:+0.00  Ez:+0.00    Radius:100.00
> .Tower angle :    Tx:+0.00  Ty:+0.00  Tz:+0.00
> Iteration : 01                                    std dev:2.665
> .Height:297.85    Ex:-0.18  Ey:-0.13  Ez:+0.00    Radius:100.68
> .Tower angle :    Tx:-0.05  Ty:+0.08  Tz:+0.00
> Iteration : 02                                    std dev:0.128
> .Height:297.77    Ex:-0.19  Ey:-0.09  Ez:+0.00    Radius:100.80
> .Tower angle :    Tx:-0.07  Ty:+0.15  Tz:+0.00
> Iteration : 03                                    std dev:0.025
> .Height:297.78    Ex:-0.17  Ey:-0.09  Ez:+0.00    Radius:100.78
> .Tower angle :    Tx:-0.09  Ty:+0.20  Tz:+0.00
> Iteration : 04                                    std dev:0.022
> .Height:297.80    Ex:-0.14  Ey:-0.07  Ez:+0.00    Radius:100.79
> .Tower angle :    Tx:-0.10  Ty:+0.22  Tz:+0.00
> Iteration : 05                                    std dev:0.019
> .Height:297.81    Ex:-0.13  Ey:-0.06  Ez:+0.00    Radius:100.80
> .Tower angle :    Tx:-0.10  Ty:+0.25  Tz:+0.00
> Calibration OK                                    rolling back.
> .Height:297.80    Ex:-0.14  Ey:-0.07  Ez:+0.00    Radius:100.79
> .Tower angle :    Tx:-0.10  Ty:+0.22  Tz:+0.00
> Save with M500 and/or copy to Configuration.h

### Actual behavior

22:27:58.623 : G33 Auto Calibrate
22:27:58.625 : Checking... AC
22:27:58.628 : .Height:280.00  Ex:+0.00  Ey:+0.00  Ez:+0.00  Radius:125.63
22:27:58.630 : .               Tx:+0.00  Ty:+0.00  Tz:+0.00
22:28:03.622 : echo:busy: processing
22:28:08.622 : echo:busy: processing
22:28:13.657 : echo:busy: processing
22:28:18.656 : echo:busy: processing
22:28:21.834 : Correct delta settings with M665 and M666"

### Steps to Reproduce

Connect to pronterface, home, send G33

### Version of Marlin Firmware

2.0.9.2

### Printer model

Custom

### Electronics

- Mks eagle - Mks mini 12864 V3 - Anycubic detachable probe.

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

[Configuration_G33.zip](https://github.com/MarlinFirmware/Marlin/files/7948729/Configuration_G33.zip)

ellensp, 2022-01-27:
Please try in verbose mode G33 V3  see it that is any more enlightening.

Voxel3DPrinting, 2022-01-27:
Still says the exact same;

"SENDING:G33 V3
G33 Auto Calibrate
Checking... AC
.Height:280.00  Ex:+0.00  Ey:+0.00  Ez:+0.00  Radius:125.63
.               Tx:+0.00  Ty:+0.00  Tz:-0.00
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Correct delta settings with M665 and M666
echo:busy: processing
echo:busy: processing"

ellensp, 2022-01-27:
Darn, that wasn't any more helpful at all...

Voxel3DPrinting, 2022-01-27:
Nope... What else can I try? Or could it really be a bug?

ellensp, 2022-01-27:
I would suspect it is a bug. 
Do you know what version worked last? Ie where you got the working log from? 
I've just gotten an old delta moving (literally 1 minute ago)   It has a Z_PROBE_ALLEN_KEY, im about to wire that up and see if I can get it to work. 
Then perhaps tomorrow onto some testing.

Voxel3DPrinting, 2022-01-27:
I have absolutely no clue which version worked last.

I've had a BLTouch before this probe, which had the same problem, although when just trying to do G33 over and over I've grotten that one to work once

ellensp, 2022-01-27:
works on 2.0.9.1  for me, 2.0.9.2 is trying but giving me grief with bogus probe error 2.0.9.3  completed but it sent something pronterface didn't like (need to re test this)

Voxel3DPrinting, 2022-01-27:
Ohkee, so If I try to get firmware to 2.0
9.0.1 it should work?

Voxel3DPrinting, 2022-01-27:
Hey,

I tried using the 2.0.9.1 marlin firmware, but it does not support my mainboard...... (Mks eagle)..
How can I try to fix it? will 2.0.9.3 work?

Enzo-Enz, 2022-01-28:
> ### Did you test the latest `bugfix-2.0.x` code?
> Yes, and the problem still exists.
>
> ### Bug Description
> hey, I have a custom delta printer and am running into issues with G33 auto calibrate..
>
> It [...]

Hey
I have the same issue 
Also my delta can not find the center of the bed!

Voxel3DPrinting, 2022-01-28:
> 

About the center of the bed; have you made sure that the nozzle to probe offset is right? and are your endstops on top level to eachother?

Enzo-Enz, 2022-01-28:
> > 
> 
> About the center of the bed; have you made sure that the nozzle to probe offset is right? and are your endstops on top level to eachother?

nozzle to probe offset(0,0,0)
and about endstops,yes I'm sure
[image]

Voxel3DPrinting, 2022-01-29:
Z offset should be -10 or something, since the probingpoint is underneth the nozzle

Voxel3DPrinting, 2022-01-29:
> Please try in verbose mode G33 V3  see it that is any more enlightening.
> 

Have you tested with 2.9.0.3??
I cannot use 2.0.9.1 because my mainboard is not supported

Enzo-Enz, 2022-01-30:
> Z offset should be -10 or something, since the probingpoint is underneth the nozzle

yeah,but till It doesn't work fine,I can't find exact Z offset!

Voxel3DPrinting, 2022-02-03:
Hi, I recompiled marlin 2.0.9.3 and G33 works good now!
I enabled Z offset wizard to find the correct z offset, might work for you to!

closed 2022-02-03 by Voxel3DPrinting