MarlinFirmware/Marlin #23633: [BUG] Part cooling fan not functioning in Marlin 2.0.9.3 on SKR 1.4 Turbo

lorroi, 2022-01-27:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Earlier this month I got around to configure Marlin 2.0.9.3 for my setup . 
I am running a Prusa Bear mod with an BigTreeTech SKR 1.4 Turbo motherboard. When I first compiled it, uploaded it to my printer and did some initial testing, I quickly realized the part cooling fan will never turn on. Not mid-print, not when idle, not when I set it through the LCD controller. I brushed it aside then, but since then had revisited the task of getting the firmware to work. 

Whatever I do, though, I cannot get the part cooling fan to turn on. This issue does not exist in version 2.0.9.2. For now I will revert to this, but I really would like to use the tachometer function introduced in 2.0.9.3.

I have it plugged in pin P2_03, which works for version 2.0.9.2

### Expected behavior

I expect the part-cooling fan to turn on when I send the M106 command, or using the LCD controller.

### Actual behavior

The part cooling fan does not turn on whatever I do

### Version of Marlin Firmware

2.0.9.3

### Printer model

Prusa bear

### Electronics

SKR 1.4 Turbo

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/7951835/config.zip)

descipher, 2022-02-01:
Please comment out this line in configuration.h 
Line 2853 #define FAN_SOFT_PWM
The part FAN_PIN is PWM capable and we need to see if it operates in PWM mode.

Seelenkind, 2022-02-02:
Swap Fan1 & Fan2 and try it. 
[image]

lorroi, 2022-02-02:
> Please comment out this line in configuration.h
> Line 2853 #define FAN_SOFT_PWM
> The part FAN_PIN is PWM capable and we need to see if it operates in PWM mode.

Good point! I will try once I got some time at hands. Thanks for the suggestion!

lorroi, 2022-02-02:
> Swap Fan1 & Fan2 and try it.
> [image]

Fan2 is directly railed to VCC in, so anytime i turn on the printer, the part cooling fan would be at 100% with no possibility to change it.

lorroi, 2022-02-02:
> Please comment out this line in configuration.h Line 2853 #define FAN_SOFT_PWM The part FAN_PIN is PWM capable and we need to see if it operates in PWM mode.

I have commented out #define FAN_SOFT_PWM and the part cooling fan is running again. Sadly, now it is will only operate above 30 % and the annoying buzzing sound is back. But at least it is operational, so it must be something in FW. Until the issue is resolved, I'll go back to 2.0.9.2 though. 
Thanks for the help! :)

descipher, 2022-02-02:
Ok, Thanks for the verify, I can now look at what's causing that state.

lorroi, 2022-02-02:
Thanks a lot!

descipher, 2022-02-03:
I have tested on 2 LCP1768 boards, they both work with FAN_SOFT_PWM and the part cooling FAN on pin P2_03.
I suspect the issue may be resolved in bugfix-2.0.x. 
Have you tested it on that version yet?

lorroi, 2022-02-06:
I have tested the bugifx branch and the problem is now fixed! Thanks for the advice!

closed 2022-02-06 by ellensp