MarlinFirmware/Marlin #23634: [BUG] Fysetc S6 Y-Stepper Pins Incorrect

lachtan86, 2022-01-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The Issue [#23044](https://github.com/MarlinFirmware/Marlin/issues/23044) was resolved incorrectly. It was correct for Board Fysetc S6 V2! Not for FysetC S6 v1.2! 

### Bug Timeline

23044

### Actual behavior

TMC Communication Error for Y Stepper only. M122 receives "Testing Y connection... Error: All LOW"

### Steps to Reproduce

Flash firmware with stock pins files
Reboot
Get TMC Connection Error displayed - M122 shows Testing Y connection... Error: All LOW

### Version of Marlin Firmware

2.0.9.3

### Additional information & file uploads

[image]

in Marlin/src/pins/stm32f4/pins_FYSETC_S6.h

```
  #ifndef Y_SERIAL_RX_PIN
    #define Y_SERIAL_RX_PIN                 PC4    <---- INCORRECT
  #endif
```
PC4 pin must be defined in pins_FYSETC_S6_V2_0.h

```
  #ifndef Y_SERIAL_RX_PIN
    #define Y_SERIAL_RX_PIN                 PE13    <---- CORRECT
  #endif
```

ellensp, 2022-01-29:
Yip your correct.  A PR created to rectify this

ellensp, 2022-01-30:
This PR has been merged into bugfix. closing.

closed 2022-01-30 by ellensp