MarlinFirmware/Marlin #23641: Fysetc Cheetah not working with STM32RC_fysetc build envs

rlagerweij, 2022-01-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When compiling recent versions of the 2.0.x branch for the Fysetc Cheetah 1.2b using the STM32F103RC_fysetc default_envs in platformio.ini it creates a firmware file which does not work on the board. The mainboard seems frozen. 

The compilation completes with no errors. Flashing completed with no errors. The screen remains blank (backlight is on) and it is not possible to connect to the printer over the serial port with either putty or pronterface. Flashing another hex file remains possible.

Tested with the latest code of today in both the 2.0.x and Bugfix-2.0.x branches.

Using git bisect I have found that the issue started with commit b6cb56f396e58b95d7e3f7750f388373bfbd01dd

That commit introduces new build_envs for STM32 targets. It also renames the old STM32F103RC_fysetc to STM32F103RC_fysetc_maple and introduces a new set of parameters under that name.

Building the exact same code and config with the default_env set to STM32F103RC_fysetc_maple results in a working firmware on the latest commits.

Issue is the same as #23067

### Bug Timeline

The bug was introduced just after the 2.0.9.1 release with commit b6cb56f396e58b95d7e3f7750f388373bfbd01dd

### Expected behavior

LCD screen shows bootlogo and can be operated from the status screen

### Actual behavior

LCD remains blank and the printer cannot be communicated with over usb(serial). Flashing remains possible.

### Steps to Reproduce

* checkout latest version of the Marlin code
* set default_envs to STM32F103RC_fysetc in platformio.ini
* copy default configs from "Merlin-Configurations/config/examples//Creality/Ender-3/FYSETC Cheetah 1.2/base" to the Marlin folder
* Build with VS code using the platform.io extension
* Flash the hex file with FlyMCU program from Fysetc Github page

### Version of Marlin Firmware

2.0.9.3 (maple)

### Printer model

Ender 3 pro

### Electronics

Fysetc Cheetah v1.2b

### Add-ons

none

### Additional information & file uploads

[broken_firmware.zip](https://github.com/MarlinFirmware/Marlin/files/7961924/broken_firmware.zip)
[working_firmware.maple.hex.zip](https://github.com/MarlinFirmware/Marlin/files/7961930/working_firmware.maple.hex.zip)

J0hnMatrix, 2022-02-13:
I can confirm this is working with _STM32F103RC_fysetc_maple_ in platformio.ini and is **NOT** working with _STM32F103RC_fysetc_

BRS3CTY, 2022-03-03:
I have the same here, only maple is working.

rlagerweij, 2022-03-09:
may be related to https://github.com/MarlinFirmware/Marlin/issues/22670

rlagerweij, 2022-03-19:
I'm happy to test things  or try to code a solution, but I just don't know where to start on this. If anyone has a hint what to look for....

BRS3CTY, 2022-03-20:
I'm trying to fix this too but i have no idea how to edit source of platformio board, i think that there is some clue.
Look at log of build, if you building STM32F103RC_fysetc build takes files from src\HAL\STM32\ , when you build STM32F103RC_fysetc_maple it takes from src\HAL\STM32F1\ and it works.
But i never could build non maple becouse of error 
Marlin\src\HAL\STM32\timers.cpp:328:40: error: static assertion failed: One or more timer conflict detected. Examine "timers_in_use" to help identify conflict.
So i can't even get blank lcd of STM32F103RC_fysetc..
I would never have tried to figure it out if not for the message in maple build:
 warning: #warning "Maple build environments are deprecated. Please use a non-Maple build environment. Report 
issues to the Marlin Firmware project."

thinkyhead, 2022-04-21:
> Examine "timers_in_use" to help identify conflict.

After a build (or attempt) in Visual Studio Code you can open the file `HAL/STM32/timers.cpp` and hover the mouse pointer over the definition of `timers_in_use` (line 304) and it should display the contents of the array in a popup. If two or more the timers are the same, that tells us where the conflict lies.