MarlinFirmware/Marlin #23647: [FR] Decouple Anycubic LCD definition from non-LCD settings.

sorgelig, 2022-01-29:
### Is your feature request related to a problem? Please describe.

Yes. If i want to change original Chiron LCD with traditional 12864 display, then i need to preserve Chiron related pin definition in `pins_TRIGORILLA_14.h`.

### Are you looking for hardware support?

Anycubic Chiron

### Describe the feature you want

add define ANYCUBIC_CHIRON in configuration.h which will be used in `pins_TRIGORILLA_14.h` and will be independent from used LCD.

thinkyhead, 2022-04-21:
The Chiron display is selected with `ANYCUBIC_LCD_CHIRON`. Other LCDs should work with this board also, which is very close to a RAMPS.

stef-ladefense, 2022-09-07:
for use with 3party uart lcd type TFTxx with trigorilla : disable anycubic lcd and add in Configuration.h
/**
 * Select a secondary serial port on the board to use for communication with the host.
 * Currently Ethernet (-2) is only supported on Teensy 4.1 boards.
 * :[-2, -1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 3       // uart3 for use interface used by old anycubic lcd
#define BAUDRATE_2 115200  // :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000] Enable to override BAUDRATE

sorgelig, 2022-09-07:
It won't work as trigorilla definition file uses chiron specific define. If you won't use chiron type then some pins will be messed.

SwiftNick, 2022-10-06:
I see what you are getting at due to the switched defs: [Chiron specific pin defs](https://github.com/MarlinFirmware/Marlin/blob/0d8a695ea96bc137059fcb88e65befc5c662adbd/Marlin/src/pins/ramps/pins_TRIGORILLA_14.h#L98) overriding endstops and filament runout pins.

If you're going to swap out the LCD you will need to disable the Chiron LCD code, but you would also be changing the pin defs to suit the new display and SD slot anyway. I was advised to add pin overrides to the configuration.h files for non-standard, but supported machine configurations. Doing it this way, reduces the number of files you actually need to change.

sorgelig, 2022-10-07:
I don't use trigorilla board anymore, so i just thought it could be useful for other users anyway.

closed 2022-10-07 by sorgelig