MarlinFirmware/Marlin #23649: [BUG]  Marlin shows the wron temperature on head0 when selected head1

gnugu13, 2022-01-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

when i heat up the heat1 and selected mit T0 head0 the rigth temperature shows an log.
but when i select head1 with T1 the temperature of the head 0 shows the same temperature like head1 (but no heating).
on the Log you can see:
with command T1:
T0:199.91 /200.00 B:23.59 /0.00 T0:22.77 /0.00 T1:199.91 /200.00 @:73 B@:0 @0:0 @1:73
 T0:200.00 /200.00 B:23.59 /0.00 T0:22.97 /0.00 T1:200.00 /200.00 @:70 B@:0 @0:0 @1:70
 T0:199.91 /200.00 B:23.48 /0.00 T0:22.81 /0.00 T1:199.91 /200.00 @:73 B@:0 @0:0 @1:73
 T0:199.97 /200.00 B:23.71 /0.00 T0:22.97 /0.00 T1:199.97 /200.00 @:71 B@:0 @0:0 @1:71
 T0:199.91 /200.00 B:23.79 /0.00 T0:22.93 /0.00 T1:199.91 /200.00 @:73 B@:0 @0:0 @1:73
with command T0:
 T0:22.93 /0.00 B:23.32 /0.00 T0:22.93 /0.00 T1:199.89 /200.00 @:0 B@:0 @0:0 @1:74
 T0:22.85 /0.00 B:23.55 /0.00 T0:22.85 /0.00 T1:200.00 /200.00 @:0 B@:0 @0:0 @1:71
 T0:23.05 /0.00 B:23.44 /0.00 T0:23.05 /0.00 T1:199.94 /200.00 @:0 B@:0 @0:0 @1:73
 T0:22.89 /0.00 B:23.55 /0.00 T0:22.89 /0.00 T1:200.07 /200.00 @:0 B@:0 @0:0 @1:69
 T0:22.97 /0.00 B:23.52 /0.00 T0:22.97 /0.00 T1:200.03 /200.00 @:0 B@:0 @0:0 @1:70
 T0:200.07 /200.00 B:23.55 /0.00 T0:22.89 /0.00 T1:200.07 /200.00 @:69 B@:0 @0:0 @1:69

whats wrong in the configuration on Marlin?

### Bug Timeline

new

### Expected behavior

I would see the correct temperature on head 0 and head 1 regardless oft the select T0 or T1.

### Actual behavior

shows the wrong temperatures im log, when select T0 and heating up head 1

### Steps to Reproduce

1. usually select head 0 (T0)
2. on my repetier host switch on head 1 to 200 °C
3. shows temperature rising
4. switch on the UI to T1
5. head 0 and head1 shows the same temperature

### Version of Marlin Firmware

Marlin-Bugfix-2.0.92.x

### Printer model

X400

### Electronics

BTT SKR V2.0 REV B

### Add-ons

TFT35 E3 V3.0

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

Repetier Host

### Additional information & file uploads

[marlin_2.0.zip](https://github.com/MarlinFirmware/Marlin/files/7966639/marlin_2.0.zip)

ellensp, 2022-01-30:
You did not test bugfix. I know this because I know that bug is fixed in bugfix. I fixed it 18 days ago!
see https://github.com/MarlinFirmware/Marlin/pull/23505

Issues is first temperature is active extruder, should be T, not T0

closed 2022-01-30 by ellensp

gnugu13, 2022-01-31:
Tnx @ellensp ,
it works now. my bugfix seems to be to old.
Regards