MarlinFirmware/Marlin #23651: Mega2560 Print Fan Problem

jwylie237, 2022-01-30:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I Have a Genuine Mega2560 powered 3dprinter using TMC2209's in UART mode on an RAMPS1.4 EFB setup. I upgraded to Marlin 2.0.9.3 and all is working well apart from a issue with the Print cooling Fan. After a print has begun and the Print Fan is running I cannot open the LCD menu without the Print Fan shutting down. It happens every time I do no matter how far into a print I am. Furthermore with the Printer in an Idle state I can navigate to the Temperature Menu and set a Print Fan speed but only once. The firmware wont accept any speed changes on the Print Fan again unless the turn the Print Fan off before setting a new speed first.

### Bug Timeline

New Bug

### Expected behavior

n

### Actual behavior

n

### Steps to Reproduce

n

### Version of Marlin Firmware

2.0.9.3

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

ellensp, 2022-01-30:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

jwylie237, 2022-01-30:
I have done so. I get the same behavior.

ellensp, 2022-01-30:
Please attach your configuration files.

jwylie237, 2022-01-31:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7967260/Configuration.zip)

descipher, 2022-01-31:
I have exactly the same config running with NO fan issues using an ATMega2560 + RAMPS_14_EFB on the very latest bugfix-2.0.x
TMC2209's in UART mode on an RAMPS1.4 EFB setup <- Not sure how UART would be possible on a board without SPI to them.

ellensp, 2022-01-31:
ramps can have TMC2209's in UART no issues. You need your own wiring. 
This user is using SLAVE_ADDRESS so must be using the same uart for all.
@jwylie237 please provide your modified Marlin/src/pins/ramps/pins_RAMPS.h

descipher, 2022-01-31:
That's what I am getting at, It has to be heavily modified.

jwylie237, 2022-01-31:
> I have exactly the same config running with NO fan issues using an ATMega2560 + RAMPS_14_EFB on the very latest bugfix-2.0.x TMC2209's in UART mode on an RAMPS1.4 EFB setup <- Not sure how UART would be possible on a board without SPI to them.

UART (Universal Asynchronous Receiver Transmitter) and SPI (Serial Peripheral Interface) are different communication methods. They are not compatible. UART sends and receives data between the Master and Slave Chip via the same data line. You set slave addresses via software or Hardware to differentiate between the Chips. SPI uses a Chip select Pin for each Chip and an input and output data line for each chip. SPI only allows you to communicate with one Chip at a time while UART lets you communicate with all the Chips simultaneously.

There is no heavy modification needed for UART on a RAMPS board. All you have to do is set Slave addresses via Hardware or Software and set the SERIAL TX and SERIAL RX pins to the same pin and off you go. The only limitation is you can only use 4 Chips on the same SERIAL RX /TX pin.

descipher, 2022-01-31:
We have no default FAN issues with PWM or FAN_SOFT_PWM on the RAMPS_14_EFB config you have posted. You can enable PINS_DEBUGGING and issue an M43 as more info and any other mods you may have in place.

jwylie237, 2022-01-31:
> ramps can have TMC2209's in UART no issues. You need your own wiring. This user is using SLAVE_ADDRESS so must be using the same uart for all. @jwylie237 please provide your modified Marlin/src/pins/ramps/pins_RAMPS.h and any other files you changed Suspect user is using hardware uart. and perhaps there is a uart/pwm conflict

[pins_RAMPS.zip](https://github.com/MarlinFirmware/Marlin/files/7967543/pins_RAMPS.zip)

ellensp, 2022-01-31:
user is using software serial on pin 65 for all 4 steppers. 
Do change level interrupts mess with timers/pwm?

descipher, 2022-01-31:
Pin 65 does not look like an issue, it is a PCINT11 vector. This issue may be a result of the LCD refresh change and the CPU load due to serial IO activity to the steppers.

M106 Pn Sn  should be tried to see if it processes the fan speed update via the parser.

descipher, 2022-01-31:
@jwylie237 What version were you running prior to 2.9.0.3 and have you tried bugfix within the last 12 hours?

jwylie237, 2022-01-31:
Bugfix is currently running on the Mega, I tried it a few hours ago and got the same results. Prior version of marlin was 2.0.9.1

descipher, 2022-01-31:
Have you checked if you can change the FAN speed via M106.

jwylie237, 2022-01-31:
I have not. I will give it a go however I do not expect anything to change.

jwylie237, 2022-01-31:
Ok. interesting results. I can set an initial Fan speed via the M106 command and then specify different speeds after that via the M106 command and the printer responds as it should. If I enter the LCD menu after the Fan is running it shuts down as I currently expect it to. Now when I issue the M106 command again the printer does not respond to it unless I command the print fan to shutdown first.

descipher, 2022-01-31:
Is a print active when this occurs?

jwylie237, 2022-01-31:
Sorry should have specified that. This was done in an Idle state.

descipher, 2022-01-31:
I can perform the same LCD based fan control operation without issue. It would be useful to see what PINS_DEBUGGING reports.

jwylie237, 2022-01-31:
Under what circumstances?

descipher, 2022-01-31:
The tool simply reports what PINs are mapped, it helps in conflict determination. There are no special circumstances.

jwylie237, 2022-01-31:
Here it is

PIN:   0   Port: E0        RXD0                                   protected
PIN:   1   Port: E1        TXD0                                   protected
PIN:   2   Port: E4        <unused/unknown>                       Input  = 0                           TIMER3B   PWM:     0    WGM: 0    COM3B: 0    CS: 2    TCCR3A: 0    TCCR3B: 2    TIMSK3: 2   non-standard PWM mode
PIN:   3   Port: E5        X_MIN_PIN                              protected
.                          X_STOP_PIN                             protected
PIN:   4   Port: G5        SERVO3_PIN                             Input  = 0    TIMER0B   PWM:     0    WGM: 3    COM0B: 3    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 3   overflow interrupt enabled
PIN:   5   Port: E3        SERVO2_PIN                             Input  = 0    TIMER3A   PWM:  -25536    WGM: 0    COM3A: 0    CS: 2    TCCR3A: 0    TCCR3B: 2    TIMSK3: 2   non-standard PWM mode   compare interrupt enabled
PIN:   6   Port: H3        SERVO1_PIN                             Input  = 1    TIMER4A   PWM:     0    WGM: 14    COM4A: 2    CS: 1    TCCR4A: 2    TCCR4B: 25    TIMSK4: 0
PIN:   7   Port: H4        <unused/unknown>                       Input  = 0                           TIMER4B   PWM:     0    WGM: 14    COM4B: 2    CS: 1    TCCR4A: 2    TCCR4B: 25    TIMSK4: 0
PIN:   8   Port: H5        HEATER_BED_PIN                         protected
.                          MOSFET_C_PIN                           protected
PIN:   9   Port: H6        FAN_PIN                                protected
.                          MOSFET_B_PIN                           protected
PIN:  10   Port: B4        HEATER_0_PIN                           protected
.                          MOSFET_A_PIN                           protected
PIN:  11   Port: B5        SERVO0_PIN                             Output = 0    TIMER1A   PWM:  2000    WGM: 4    COM1A: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode   compare interrupt enabled
PIN:  12   Port: B6        <unused/unknown>                       Input  = 0                           TIMER1B   PWM:     0    WGM: 4    COM1B: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
PIN:  13   Port: B7        LED_PIN                                Input  = 0    TIMER0A   PWM:   128    WGM: 3    COM0A: 3    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 3   compare interrupt enabled   overflow interrupt enabled
 .                  TIMER1C is also tied to this pin                  TIMER1C   PWM:     0    WGM: 4    COM1C: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
PIN:  14   Port: J1        Y_MIN_PIN                              protected
.                          Y_STOP_PIN                             protected
PIN:  15   Port: J0        <unused/unknown>                       Input  = 0
PIN:  16   Port: H1        LCD_PINS_RS                            Output = 1
PIN:  17   Port: H0        LCD_PINS_ENABLE                        Output = 0
PIN:  18   Port: D3        Z_MIN_PIN                              protected
.                          Z_STOP_PIN                             protected
PIN:  19   Port: D2        <unused/unknown>                       Input  = 0
PIN:  20   Port: D1        <unused/unknown>                       Input  = 1
PIN:  21   Port: D0        <unused/unknown>                       Input  = 1
PIN:  22   Port: A0        <unused/unknown>                       Input  = 1
PIN:  23   Port: A1        LCD_PINS_D4                            Output = 0
PIN:  24   Port: A2        E0_ENABLE_PIN                          protected
PIN:  25   Port: A3        LCD_PINS_D5                            Output = 0
PIN:  26   Port: A4        E0_STEP_PIN                            protected
PIN:  27   Port: A5        LCD_PINS_D6                            Output = 0
PIN:  28   Port: A6        E0_DIR_PIN                             protected
PIN:  29   Port: A7        LCD_PINS_D7                            Output = 0
PIN:  30   Port: C7        E1_ENABLE_PIN                          Input  = 1
PIN:  31   Port: C6        BTN_EN1                                Input  = 1
PIN:  32   Port: C5        <unused/unknown>                       Input  = 1
PIN:  33   Port: C4        BTN_EN2                                Input  = 1
PIN:  34   Port: C3        E1_DIR_PIN                             Input  = 0
PIN:  35   Port: C2        BTN_ENC                                Input  = 1
PIN:  36   Port: C1        E1_STEP_PIN                            Input  = 0
PIN:  37   Port: C0        BEEPER_PIN                             Output = 0
PIN:  38   Port: D7        X_ENABLE_PIN                           protected
PIN:  39   Port: G2        <unused/unknown>                       Input  = 0
PIN:  40   Port: G1        E_MUX0_PIN                             Input  = 1
PIN:  41   Port: G0        KILL_PIN                               Input  = 1
PIN:  42   Port: L7        E_MUX1_PIN                             Input  = 0
PIN:  43   Port: L6        <unused/unknown>                       Input  = 0
PIN:  44   Port: L5        E_MUX2_PIN                             Input  = 0    TIMER5C   PWM:     0    WGM: 14    COM5C: 0    CS: 1    TCCR5A: 2    TCCR5B: 25    TIMSK5: 0
.                          E1_CS_PIN                              Input  = 0    TIMER5C   PWM:     0    WGM: 14    COM5C: 0    CS: 1    TCCR5A: 2    TCCR5B: 25    TIMSK5: 0
[... cut ...]
.                          Z_SERIAL_RX_PIN                        Output = 1
.                          E0_SERIAL_TX_PIN                       Output = 1
.                          E0_SERIAL_RX_PIN                       Output = 1
PIN:  66   Port: K4 (A12)  TEMP_0_CS_PIN                          Input  = 1
PIN:  67   Port: K5 (A13)  TEMP_0_PIN                             protected
PIN:  68   Port: K6 (A14)  TEMP_BED_PIN                           protected
PIN:  69   Port: K7 (A15)  TEMP_1_PIN                             Analog in =  1023

descipher, 2022-01-31:
I don't see any blaring pin conflicts.
Please comment out:
#define SPEAKER
and report if the behavior changes.

jwylie237, 2022-01-31:
Turning off the Speaker solved the issue, I now have full control over the Print Fan in both idle and printing states.

jwylie237, 2022-01-31:
And it does not shutdown when I open the LCD Menu

descipher, 2022-01-31:
Ok, Thanks for the verify, looks like there are some dangerous behaviors in the tone code that are touching system timers.

descipher, 2022-01-31:
@thinkyhead FYI marlin/src/lib/buzzer.cpp uses Arduino calls that can directly manipulate system timers and interrupts which may cause instability on AVR platforms.

borland1, 2022-01-31:
As a workaround, is it possible to reassign Timers in `Configuration.h` that are used for fan PWM?

I have an open a bug report ( #21109 ) for similar fan problem, where there is a timer PWM glitch that appears whenever pressing the rotary encoder button.  The the` AUTO_FAN` ( extruder fan ) power is interrupted for a second or two, but then resumes running.

thinkyhead, 2022-01-31:
> FYI marlin/src/lib/buzzer.cpp uses Arduino calls that can directly manipulate system timers and interrupts which may cause instability on AVR platforms. 

The timer used for Tone has been accounted for in past iterations of Marlin, so what is now in conflict? Last we checked, Timer 2 was being assigned for Tone on AVR.

descipher, 2022-01-31:
// search for an unused timer.
  for (int i = 0; i < AVAILABLE_TONE_PINS; i++) {
    if (tone_pins[i] == 255) {
      tone_pins[i] = _pin;
      _timer = pgm_read_byte(tone_pin_to_timer_PGM + i);
      break;
    }
  }

As far as I understand the code it gets to pick the timer.

descipher, 2022-01-31:
With #define SPEAKER it was hammering FAN_PIN D9 PWM settings, OC2B. But yes your correct it should be restricted to timer 2 on the RAMPS. Will need to consider a method of detecting it's in use.

descipher, 2022-01-31:
> As a workaround, is it possible to reassign Timers in `Configuration.h` that are used for fan PWM?

Timers are associated with PINs and 8 bit AVRs do not have alternate pin maps. Just comment out #define SPEAKER or use SOFT_FAN_PWM.

borland1, 2022-01-31:
If I comment out #define SPEAKER, the AUTO_FAN (extruder fan) glitch associated with the rotary encoder button being pressed.

I don't see a way to enable SOFT_FAN_PWM in the configs.  Is this possible with AUTO_FAN?

descipher, 2022-01-31:
Just checked. Nope, will not work with AUTO_FAN, it will detect the PWM pin and use that mode. Otherwise it would be on or off with a digital pin.
Just run it without tones, you still get beeps on the buzzer, just not so elegant as tones ..

closed 2022-02-11 by ellensp