MarlinFirmware/Marlin #23652: [BUG] PID values for bed and hotend in Firmware is not what is reported by M503

Smitty-Werbenjagermenjensen, 2022-01-31:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After doing a PID tune for the bed and hotend, I hard code the PID kp, kd, and ki values for both the hotend and bed in the configuration.h file. 

I was having some problems with the printer shutting off due to a temperature problem with the bed. I issued an M503 command to take a look at the PID values for the hotend and bed. I noticed that it was 0.00 for both the hotend and bed for Ki, Kp, and Kd. 

Screenshot: 
[image]

In code, I have defined the following PID values for the bed: 

[image]

PID values for the hotend: 

[image]

### Bug Timeline

I just noticed it. I have previously applied PID values in this way. I thought it would apply. Not sure if the same thing happened before. 

### Expected behavior

I would expect the hard coded PID values to be used by the firmware. 

### Actual behavior

M503 output shows 0.00.

### Version of Marlin Firmware

Marlin 2.0.9.3

### Printer model

Creality Ender 3

### Electronics

BTT SKR Mini e3 V3.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7967418/Configuration.zip)

Smitty-Werbenjagermenjensen, 2022-01-31:
I have very often been receiving the message for heating failed after hard coding those PID values for the bed. As I mentioned, even though I hard coded those values, M503 was showing 0.00. 

I have now set the PID using M304 and M301. I am no longer receiving the heating failed error message. 

Before manually setting it with M304 and M301, the bed would struggle around 50 Celsius, and would not exceed 50 Celsius, when the target was 60 Celsius. Now it can very swiftly reach 60 Celsius. 

I think it shows that hard coding the PID values does not work at all. This must be a bug.

thisiskeithb, 2022-01-31:
Remember to reset your EEPROM after flashing with `M502` followed by `M500`.

ellensp, 2022-02-01:
Eeprom values take precedence. As already mentioned "emember to reset your EEPROM after flashing with M502 followed by M500."

closed 2022-02-01 by ellensp