MarlinFirmware/Marlin #23653: [BUG] (bug summary)

Smitty-Werbenjagermenjensen, 2022-01-31:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When going to the LCD Menu -> Temperature -> Preheat PLA, then none of "Preheat PLA", "Preheat PLA End", "Preheat PLA Bed" options work. 

When clicking on one of those options, it takes me back to the main information screen. But the temperature does not change, and the target temperature is still 0 on top of the bed and hotend icons. 

### Bug Timeline

Just noticed it. Just upgraded to the SKR Mini e3 V3.0. 

### Expected behavior

I would expect the bed, hotend or both to heat up according to the function of that feature. 

### Actual behavior

The temperature does not change at all. 

### Version of Marlin Firmware

Marlin 2.0.9.3

### Printer model

Creality Ender 3

### Electronics

SKR Mini e3 V3.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7967494/Configuration.zip)

thisiskeithb, 2022-01-31:
Remember to reset your EEPROM after flashing with `M502` followed by `M500`.

Smitty-Werbenjagermenjensen, 2022-02-03:
Thanks. `M502` followed by `M500` fixed the problem.

closed 2022-02-03 by Smitty-Werbenjagermenjensen