MarlinFirmware/Marlin #23654: [BUG] Several values in "Advanced Settings" menu are incorrectly showing 0.00.

Smitty-Werbenjagermenjensen, 2022-01-31:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When going to Configuration -> Advanced Settings, menus such as: 

- Probe Offsets
- Junction Deviation
- Steps/mm
- Acceleration
- Velocity

are all showing 0.00 for all of the values. When doing M503, the output similarly shows 0.00 for many parameters. 

Perhaps there are more menus than what I listed above having the same problem. These are just some that I looked through. 

This is similar to the other issue I posted : #23652

### Bug Timeline

Just upgraded to SKR Mini e3 V3.0

### Expected behavior

I would expect the programmed settings to take effect that were programmed in the configuration files. 

Things like acceleration, probe offsets, junction deviation, steps/mm etc. should all have a value, needless to say. 

### Actual behavior

All of the menus and the output of M503 are showing 0.00

### Steps to Reproduce

1. Open menu
2. Take a look at Values
3. Run M503

### Version of Marlin Firmware

Marlin 2.0.9.3

### Printer model

Creality Ender 3

### Electronics

SKR Mini e3 V3.0

### Add-ons

3DTouch, bed springs, metal extruder 

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7967545/Configuration.zip)

descipher, 2022-01-31:
#define POWER_LOSS_RECOVERY can be a problem

Smitty-Werbenjagermenjensen, 2022-01-31:
I have commented #define POWER_LOSS_RECOVERY, recompiled, and updated the printer. 

All of the settings are still showing 0.00. Like everything pretty much.

descipher, 2022-01-31:
What happens if you set one and save it?

thisiskeithb, 2022-01-31:
Remember to reset your EEPROM after flashing with `M502` followed by `M500`.

Smitty-Werbenjagermenjensen, 2022-01-31:
@thisiskeithb After running M502 and then M500, the settings were all populated in the advanced configuration!

@descipher So before the above solution, if I changed a value manually, such as M301 for hotend PID values, and then did M500, that value individually would then show instead of 0.00. The other values that I did not manually change, still only showed up as 0.00.

closed 2022-01-31 by thisiskeithb

descipher, 2022-01-31:
#define EEPROM_AUTO_INIT    // Init EEPROM automatically on any errors.
  #define EEPROM_INIT_NOW   // Init EEPROM on first boot after a new build.
Are we guessing on config items, please start with only the settings you require and then add additional after items.