MarlinFirmware/Marlin #23656: manual_feedrate_mm_s undefined in marlin

guitarplayr2, 2022-01-31:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

[image]
keep getting this, no matter what i do. ive tried various different versions of marlin 2.0, none work.

### Bug Timeline

Apparently old issue, but all of the times people have had it those suggested fixes havent worked

### Expected behavior

I expected the files to compile to a .bin file that I could upload to mr printer (an ender 3 pro with a creality 4.2.2 board)

### Actual behavior

I got an error when i tried to compile, saying that manual_feedrate_mm_s was undefined

### Steps to Reproduce

Download /marlin/lcd/marlinui.cpp file

Try running in platform io in vs code

See if you get the same problem, if you do then need help fixing

i will include my files here
[Uploading Marlin-2.0.9.3 (2).zip…]()

### Version of Marlin Firmware

Currently 1.0.1, trying to upgrade to 2.0.9, got stuck here

### Printer model

Creality Ender 3 Pro with 4.2.2 board

### Electronics

Stock

### Add-ons

Added an ender extender 500 z height, need to change that in firmware, thats why i have a modified config file

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

borland1, 2022-01-31:
Try uploading your config files again.  I tried to download them unsuccessfully.  

Did you search for similar issues?  Are you sure you have your display configured correctly?  Closed report #21598 seem to be a similar issue.

ellensp, 2022-01-31:
In bugfix, you need to use bugfix config
In particular In Configuration_adv.h this will no longer work
```
// @section lcd

#if ANY(HAS_LCD_MENU, EXTENSIBLE_UI, HAS_DWIN_E3V2)

```
It should now be
```
// @section lcd

#if HAS_MANUAL_MOVE_MENU

```

isarrider, 2022-01-31:
I have the same issue and the the added lines dont work f me...

thisiskeithb, 2022-01-31:
Cannot reproduce.

Also, it's clear from your screenshot that you are not using the latest bugfix.

closed 2022-01-31 by thisiskeithb

descipher, 2022-01-31:
Not a bug, just a define change. Start with the included repo config files and then add your settings.

isarrider, 2022-02-03:
it is helpful if you use the right conf files (matching the branch u wanna compile ;))...

nspillson, 2022-02-04:
I had the same problem until I switched to the release-2.0.9.3 branch from the Configuration and grabbed the updated config from there. 

The default branch on Configuration import-2.0.x which is what might be messing you up.

As a sanity check I did a build with a clean repo so I knew it had to be something wrong in the config I was applying. Hope that helps.

tablatronix, 2022-03-05:
This is so damn confusing...  Yes I had to use release-2.0.9.3, perhaps a new master branch should be defined...

Microsparky, 2022-04-19:
I commented out the following `#error`s since it was not clear what was meant:
```
// #error "Don't build with import-2.0.x configurations!"
// #error "Use the 'bugfix...' or 'release...' configurations matching your Marlin version."
```

Something like the following would be clearer for newcomers:
```
#error "Ensure used configuration file version matches Marlin version."
#error "Use the matching 'bugfix...' or 'release...' branch from https://github.com/MarlinFirmware/Configurations."
#error "Don't build with import-2.0.x configurations!"
```

Checking and warning the user at compile time would be an enhancement!

thisiskeithb, 2022-04-20:
> Checking and warning the user at compile time would be an enhancement!

Those `#error` lines *are* the compile time check.

Don’t use configs from the `import-2.0.x` branch.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

xbral, 2022-05-18:
> 

In my case, I do this: add HAS_MANUAL_MOVE_MEN  in if clause
# section lcd

#if ANY(HAS_LCD_MENU, EXTENSIBLE_UI, HAS_DWIN_E3V2, HAS_MANUAL_MOVE_MENU)
  #define MANUAL_FEEDRATE { 50*60, 50*60, 4*60, 2*60 } // (mm/min) Feedrates for manual moves along X, Y, Z, E from panel

With this the build get success

Microsparky, 2022-05-18:
`#error` is not a check. I mean an actual compile time check like `#if`or `assert()`

thisiskeithb, 2022-05-18:
> `#error` is not a check. I mean an actual compile time check like `#if`or `assert()`

We have other checks in place, but the errors added to configs in the `import-2.0.x` branch immediately halt the build at compile time. If you are just going to bypass those, then adding more ways to prevent building won't do us any good.

**Read what the error message says** and "_Don't build with import-2.0.x configurations! Use the 'bugfix...' or 'release...' configurations matching your Marlin version._":

[image]

Microsparky, 2022-05-18:
I am trying to help here... 

The reason people (including me) are ending up here is that **It's not clear what the error means** if the user doesn't know how the project is supposed to be built. 

That is why I suggested that the `#error` is written in a way that is clear to the uninitiated.

cimba007, 2022-05-22:
Btw .. using Marlin Release 2.0.9.3 with the released config files ## https://github.com/MarlinFirmware/Configurations/archive/release-2.0.9.3.zip

```
C:\Users\USER\Downloads\Marlin-2.0.x\Marlin\src\lcd\marlinui.cpp: In static member function 'static void ManualMove::task()':
C:\Users\USER\Downloads\Marlin-2.0.x\Marlin\src\lcd\marlinui.cpp:796:60: error: 'manual_feedrate_mm_s' was not declared in this scope
         const feedRate_t fr_mm_s = (axis < LOGICAL_AXES) ? manual_feedrate_mm_s[axis] : XY_PROBE_FEEDRATE_MM_S;
                                                            ^~~~~~~~~~~~~~~~~~~~
C:\Users\USER\Downloads\Marlin-2.0.x\Marlin\src\lcd\marlinui.cpp:796:60: note: suggested alternative: 'feedrate_mm_s'
         const feedRate_t fr_mm_s = (axis < LOGICAL_AXES) ? manual_feedrate_mm_s[axis] : XY_PROBE_FEEDRATE_MM_S;
                                                            ^~~~~~~~~~~~~~~~~~~~
                                                            feedrate_mm_s
"C:\\Users\\USER\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega1284p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_SANGUINO -DARDUINO_ARCH_AVR "-IC:\\Users\\USER\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.5\\cores\\arduino" "-IC:\\Users\\USER\\AppData\\Local\\Arduino15\\packages\\anet\\hardware\\avr\\1.0.2\\variants\\sanguino" "-IC:\\Program Files (x86)\\Arduino\\libraries\\LiquidCrystal\\src" "C:\\Users\\MARCRE~1\\AppData\\Local\\Temp\\arduino_build_738809\\sketch\\src\\lcd\\dogm\\u8g_fontutf8.cpp" -o "C:\\Users\\MARCRE~1\\AppData\\Local\\Temp\\arduino_build_738809\\sketch\\src\\lcd\\dogm\\u8g_fontutf8.cpp.o"
Bibliothek LiquidCrystal in Version 1.0.7 im Ordner: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Anet V1.0.
```

ellensp, 2022-05-22:
@cimba007 
configs https://github.com/MarlinFirmware/Configurations/archive/release-2.0.9.3.zip 
with 
Marlin https://github.com/MarlinFirmware/Marlin/archive/refs/tags/2.0.9.3.zip  
works as expected

But if you use marlin https://github.com/MarlinFirmware/Marlin/archive/2.0.x.zip  it breaks
2.0.x is the latest release (ie 2.0.9.3) with selected bugfixes applied. In this instance It also includes some changes to Configuration_adv.h tests .

It is very much a version issue, config files for 2.0.9.3 need to be updated if your using 2.0.x code

cimba007, 2022-05-22:
@ellensp Please do the following:

1. Download Marlin 2.0.9.3 from the link you provided
2. Download Configuration 2.0.9.3 from the link you provided
3. Replace Configuration files with config -> examples -> Anet -> A8

.. this produces the error I posted .. the config files "should" be compatible but it seems the Anet A8 config files contains the error that @xbral mentioned.

So this IS NOT an version issue in this case and it does not "works as expected" as you said.

ellensp, 2022-05-22:
@cimba007 I already did this, before posting. IT DOES NOT ERROR

cimba007, 2022-05-22:
I think I found the confusion .. i got mixed up with alle the links .. but what is a fact:

On https://marlinfw.org/meta/download/ the download + config files DO NOT MATCH .. the lable suggests 2.0.9.3 but you will download 2.0.x

thisiskeithb, 2022-05-22:
I just verified that no errors occur if using direct links to `2.0.9.3` firmware & config releases.

```prolog
Archiving .pio/build/sanguino1284p/libFrameworkArduino.a
Indexing .pio/build/sanguino1284p/libFrameworkArduino.a
Linking .pio/build/sanguino1284p/firmware.elf
Checking size .pio/build/sanguino1284p/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  25.5% (used 4185 bytes from 16384 bytes)
Flash: [========= ]  85.3% (used 108314 bytes from 126976 bytes)
Building .pio/build/sanguino1284p/firmware.hex
============================== [SUCCESS] Took 29.71 seconds ==============================

Environment    Status    Duration
-------------  --------  ------------
sanguino1284p  SUCCESS   00:00:29.707
=============================== 1 succeeded in 00:00:29.707 ===============================

Terminal will be reused by tasks, press any key to close it.
```

Don't use the current `2.0.x` branch with `2.0.9.3` configs or you will get the `'manual_feedrate_mm_s' was not declared in this scope` error. This will be resolved once `2.0.9.4`/`2.1` are tagged & released.

tekmouse, 2022-05-24:
> In my case, I do this: add HAS_MANUAL_MOVE_MEN in if clause
>
> # section lcd
> #if ANY(HAS_LCD_MENU, EXTENSIBLE_UI, HAS_DWIN_E3V2, HAS_MANUAL_MOVE_MENU) #define MANUAL_FEEDRATE { 50_60, 50_60, 4_60, [...]

This fixed it for me. I was using the correct config.h After this add it worked. THank You!

JiffiPop, 2022-05-30:
> In my case, I do this: add HAS_MANUAL_MOVE_MEN in if clause
> # section lcd
>
> #if ANY(HAS_LCD_MENU, EXTENSIBLE_UI, HAS_DWIN_E3V2, HAS_MANUAL_MOVE_MENU) #define MANUAL_FEEDRATE { 50_60, 50_60, 4_60, [...]

Excellent work! This worked for me, too. 👍

ChrisGill77, 2022-06-01:
I have exactly the same problem. In my case, I downloaded 2.0.9.3 source and config and confirmed the version by looking in both config files. However, I wanted the files from Configurations-release-2.0.9.3>config>examples>Creality>Ender5>BigTreeTech SKR Mini E3 2.0. Copying these files and compiling led to the error.

Many thanks to XBRAL for his solution - adding HAS_MANUAL_MOVE_MENU fixed it for me too.

I had no trouble with 2.0.8.2 so I wonder if some of the 2.0.9.3 updates haven't propogated down to all the example config files.

thisiskeithb, 2022-06-02:
>  I wonder if some of the 2.0.9.3 updates haven't propogated down to all the example config files.

The download links on marlinfw.org's download page point to the `2.0.x` branch of Marlin instead of the tagged `2.0.9.3` release & the tagged `2.0.9.3` configs.

## These are currently not compatible with each other.

This will be resolved when `2.0.9.4` and `2.1` are tagged & released. In the meantime, you will need to use:

[`2.0.9.3` tagged release](https://github.com/MarlinFirmware/Marlin/archive/refs/tags/2.0.9.3.zip) with [`2.0.9.3` tagged configs](https://github.com/MarlinFirmware/Configurations/archive/refs/heads/release-2.0.9.3.zip)

...or....

[`bugfix-2.0.x` "nightly"](https://github.com/MarlinFirmware/Marlin/archive/refs/heads/bugfix-2.0.x.zip) with [`bugfix-2.0.x` configs](https://github.com/MarlinFirmware/Configurations/archive/refs/heads/bugfix-2.0.x.zip).