MarlinFirmware/Marlin #23662: MKS Robin Nano V2. TFT_LVGL_UI. TFT_ROTATION TFT_ROTATE_180

altiber, 2022-02-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

TFT_ROTATION TFT_ROTATE_180 for MKS Robin Nano V2 with TFT_LVGL_UI rotates only image and not rotate touch.

### Steps to Reproduce

Just try it on latest bugfix

### Version of Marlin Firmware

CONFIGURATION_H_VERSION 02000903

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7979373/Marlin.zip)

ellensp, 2022-02-01:
The touch screen is a physically separate device. 

run https://marlinfw.org/docs/gcode/M995.html and reconfigure it

ellensp, 2022-02-01:
I don't consider this  bug. This is the expected behavior.

altiber, 2022-02-01:
Calibration is ok. The crosses appear in the right places, but after the calibration is over, the problem remains

altiber, 2022-02-01:
And what about #define TFT_COLOR_UI? With this define there is no such problem

altiber, 2022-02-01:
And M995 video

https://user-images.githubusercontent.com/74643876/151996495-f766d918-34b5-404b-8fd9-2e32e4597c2a.mp4

altiber, 2022-04-03:
Sorry, but nobody no ideas?

closed 2022-06-13 by github-actions[bot]