MarlinFirmware/Marlin #23663: [FR] (feature summary)

jpslootmans, 2022-02-01:
### Is your feature request related to a problem? Please describe.

I installed E3V2-BLTouch-3x3-HS-v4.2.2-v2.0.1.bin firmware on my Ender V2. The BL Touch probe sometimes fails (only about once in 30 measurements). When that happens the printer stops the level measurement but starts printing the model anyway. This results in crashing the nozzle into the bed or printing in the air above the bed.

### Describe the feature you want

It would be much more convenient if the failed probing action would result in another try or tries (maybe even 3 or 4 four times). If a succesful measurement occurs, the leveling process can proceed. If the error persists the print should be aborted instead of started. 

### Additional context

I have tried to avoid failure of the probe by adding the 10 kOhm resistor to the 3.3 V as suggested in a forum, but it still happens irregularly.

thisiskeithb, 2022-02-01:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

jpslootmans, 2022-02-01:
I am sorry, but I think this is a "development-related issue", because the software is not behaving as logical as would be expected and as far as I know there is nothing the user can do about it (besides asking the developer to improve it).
Kindest regards,
JP