MarlinFirmware/Marlin #23664: [BUG] MKS robin nano v3.1 error in compiling SENSORLESS_PROBING in a COREXY with tmc2209

xamxamxam, 2022-02-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I try to enable SENSORLESS_PROBING in a corexy auto build 3d printer.
SENSORLESS_HOMING is working.

I use
// #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN // test probe max
// #define Z_MIN_PROBE_PIN PC8 // Pin 32 is the RAMPS default

but i got error in compiling :
[{
	"resource": "/c:/Users/Massimo/Documents/PlatformIO/Projects/Marlin-bugfix-2.0.x/Marlin/src/module/endstops.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'Z_MIN_PROBE_PIN' was not declared in this scope; did you mean 'Z_MIN_PROBE'?",
	"startLineNumber": 336,
	"startColumn": 26,
	"endLineNumber": 336,
	"endColumn": 26
},{
	"resource": "/c:/Users/Massimo/Documents/PlatformIO/Projects/Marlin-bugfix-2.0.x/Marlin/src/module/endstops.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'Z_MIN_PROBE_PIN' was not declared in this scope; did you mean 'Z_MIN_PROBE'?",
	"startLineNumber": 728,
	"startColumn": 26,
	"endLineNumber": 728,
	"endColumn": 26
},{
	"resource": "/c:/Users/Massimo/Documents/PlatformIO/Projects/Marlin-bugfix-2.0.x/Marlin/src/module/probe.h",
	"owner": "cpp",
	"severity": 8,
	"message": "'Z_MIN_PROBE_PIN' was not declared in this scope; did you mean 'Z_MIN_PROBE'?",
	"startLineNumber": 43,
	"startColumn": 35,
	"endLineNumber": 43,
	"endColumn": 35
}]

### Expected behavior

compilation OK!

### Version of Marlin Firmware

2.0.9.3 bugfix

### Printer model

corexy

### Electronics

mks robin nano v3

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7981760/Configuration.zip)

Bob-the-Kuhn, 2022-02-04:
> I use
> // #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN // test probe max
> // #define Z_MIN_PROBE_PIN PC8 // Pin 32 is the RAMPS default

The compile problem is because of the leading `//` in the line **`// #define Z_MIN_PROBE_PIN PC8 // Pin 32 is the RAMPS default`**.   Everything after a `//` is a comment and is ignored by the compiler.

Once that correction is made then it'll compile with either of the USB build environments.

closed 2022-02-04 by Bob-the-Kuhn

xamxamxam, 2022-02-04:
thank you for your  answer
so , in my motherboard Z_MIN in PC8 i can use
 #define Z_MIN_PROBE_PIN PC8 
and Marlin set the TMC of Z configured to send stallguard error when is probing
Correct?