MarlinFirmware/Marlin #23667: [FR] (Autoload Filament - No Runout Before Insert)

ETE-Design, 2022-02-02:
### Describe the feature you want

For people with Direct Drive Extruder mabye also Bowden it would be great to have a feature like:
"#define FILAMENT_CHANGE_RESUME_ON_INSERT"

But as an extra option have something like "First Time Preheat", so if none of the Filament Runout Sensor request

" *  1. Running a print job from media started with M24.
  *  2. The Print Job Timer has been started with M75.
  *  3. The heaters were turned on and PRINTJOB_TIMER_AUTOSTART is enabled."

is achieved then, when filament is inserted:

- Load X amount of Filament
- Preheat
- Purge

sarvenn, 2022-02-15:
Nice request. I thought prusa mk3 had a similar feature.

ETE-Design, 2022-02-15:
@sarvenn They have :-) Just need someone with the knowlage to transform it to Marlin 2.0

sarvenn, 2022-02-17:
After insertion there should be a parametere to adjust the delay time for seqeunce to start.

ETE-Design, 2022-02-19:
@sarvenn Agreed, but first we need to find a person who is willing to make the code, or try to merge the code I linked to...

sarvenn, 2022-02-22:
I hope a developer will take care of this feature addition request.