MarlinFirmware/Marlin #23677: [BUG] (bug summary)

Boldie, 2022-02-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have a problem, the homing of the z axis is not working on my Tronxy X5SA printer on latest bugfix-2.0.x. The printer starts with a safe leveling down, then does the homing for X and Y axis correctly and afterwards it moves the bed down and down again. That was it. 

I got it to work sometimes, but it was not reproducible. After moving around the z offset and moving up / down it sometimes starts to work once, so the end probe is working properly.

I had to alter the configs (set inverse direction for X/Y axis) for my Tronxy X5SA, it seems to be a special model, because it is a H-Bot.

I have attached the homing logs and my configuration below. Let me know if you need more information or if you have any tips how to dig in further.

### Bug Timeline

Today

### Expected behavior

Moving the bed upwards and doing a homing using the probe.

### Actual behavior

Bed moves down every time I did a homing and next time down again ... till to the end of the housing.

### Steps to Reproduce

Perform a Homing on GUI or  serial line.

### Version of Marlin Firmware

echo:M115 FIRMWARE_NAME:Marlin bugfix-2.0.x (Feb  4 2022 23:26:16) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Tronxy X5SA EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff Cap:SERIAL_XON_XOFF:0 Cap:BINARY_FILE_TRANSFER:0 Cap:EEPROM:1 Cap:VOLUMETRIC:1 Cap:AUTOREPORT_POS:0 Cap:AUTOREPORT_TEMP:1 Cap:PROGRESS:0 Cap:PRINT_JOB:1 Cap:AUTOLEVEL:1 Cap:RUNOUT:1 Cap:Z_PROBE:1 Cap:LEVELING_DATA:1 Cap:BUILD_PERCENT:0 Cap:SOFTWARE_POWER:0 Cap:TOGGLE_LIGHTS:0 Cap:CASE_LIGHT_BRIGHTNESS:0 Cap:EMERGENCY_PARSER:0 Cap:HOST_ACTION_COMMANDS:0 Cap:PROMPT_SUPPORT:0 Cap:SDCARD:1 Cap:REPEAT:0 Cap:SD_WRITE:1 Cap:AUTOREPORT_SD_STATUS:0 Cap:LONG_FILENAME:1 Cap:LFN_WRITE:0 Cap:CUSTOM_FIRMWARE_UPLOAD:0 Cap:EXTENDED_M20:1 Cap:THERMAL_PROTECTION:1 Cap:MOTION_MODES:0 Cap:ARCS:1 Cap:BABYSTEPPING:1 Cap:CHAMBER_TEMPERATURE:0 Cap:COOLER_TEMPERATURE:0 Cap:MEATPACK:0 Cap:CONFIG_EXPORT:0 ok

### Printer model

Tronxy X6SA

### Electronics

Stock electronics

### Add-ons

Exchanged the  bed leveling sensor to the black type; with original firmware it was working perfectly.

### Additional information & file uploads

```
G28 Z
echo:G28 Z
>>> G28  X205.00 Y165.00 Z0.00
Machine Type: CoreCartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X-40.00 Y0.00 Z0.00 (Left-Center & Same Z as Nozzle)
Auto Bed Leveling: BILINEAR (disabled)

Raise Z (before homing) by 10.00
>>> do_blocking_move_to  X205.00 Y165.00 Z0.00
>  X205.00 Y165.00 Z10.00
echo:busy: processing
<<< do_blocking_move_to  X205.00 Y165.00 Z10.00
>>> home_z_safely  X205.00 Y165.00 Z10.00
  current_position= X205.00 Y165.00 Z10.00 : sync_plan_position
  destination= X205.00 Y165.00 Z0.00 : home_z_safely
>>> do_blocking_move_to  X205.00 Y165.00 Z10.00
>  X205.00 Y165.00 Z10.00
<<< do_blocking_move_to  X205.00 Y165.00 Z10.00
>>> homeaxis(Z)
  current_position= X205.00 Y165.00 Z10.00 : Probe::set_deployed
deploy: 1
Probe::do_z_raise(10.00)
>>> do_blocking_move_to  X205.00 Y165.00 Z10.00
>  X205.00 Y165.00 Z10.00
<<< do_blocking_move_to  X205.00 Y165.00 Z10.00
>>> do_blocking_move_to  X205.00 Y165.00 Z10.00
>  X205.00 Y165.00 Z10.00
<<< do_blocking_move_to  X205.00 Y165.00 Z10.00
Home Fast: -600.00mm
>>> do_homing_move  X205.00 Y165.00 Z10.00
...(Z, -600.00, [4.00])
<<< do_homing_move  X205.00 Y165.00 Z10.00
Move Away: 5.00mm
>>> do_homing_move  X205.00 Y165.00 Z10.00
...(Z, 5.00, 4.00)
echo:busy: processing
<<< do_homing_move  X205.00 Y165.00 Z10.00
Re-bump: -10.00mm
>>> do_homing_move  X205.00 Y165.00 Z10.00
...(Z, -10.00, 2.00)
<<< do_homing_move  X205.00 Y165.00 Z10.00
>>> set_axis_is_at_home(Z)
*** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
> probe.offset.z = 0.00
Axis Z home_offset = 0.00 position_shift = 0.00
> home_offset[Z] = 0.00
  current_position= X205.00 Y165.00 Z0.00 : 
<<< set_axis_is_at_home(Z)
  current_position= X205.00 Y165.00 Z0.00 : sync_plan_position
  current_position= X205.00 Y165.00 Z0.00 : > AFTER set_axis_is_at_home
  current_position= X205.00 Y165.00 Z0.00 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to  X205.00 Y165.00 Z0.00
>  X205.00 Y165.00 Z0.00
<<< do_blocking_move_to  X205.00 Y165.00 Z0.00
<<< homeaxis(Z)
<<< home_z_safely  X205.00 Y165.00 Z0.00
  current_position= X205.00 Y165.00 Z0.00 : sync_plan_position
X:205.00 Y:165.00 Z:0.00 E:0.00 Count A:29600B:3200 Z:0
<<< G28  X205.00 Y165.00 Z0.00
ok

----- A full process; after fiddling around with z offset; also not working ----

>>> G28  X205.00 Y165.00 Z0.97
Machine Type: CoreCartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X-40.00 Y0.00 Z-0.97 (Left-Center & Below Nozzle)
Auto Bed Leveling: BILINEAR (disabled)

Raise Z (before homing) by 10.00
>>> do_blocking_move_to  X205.00 Y165.00 Z0.97
>  X205.00 Y165.00 Z10.00
echo:busy: processing
<<< do_blocking_move_to  X205.00 Y165.00 Z10.00
>>> homeaxis(X)
Home Fast: -495.00mm
>>> do_homing_move  X205.00 Y165.00 Z10.00
...(X, -495.00, [50.00])
echo:busy: processing
echo:busy: processing
<<< do_homing_move  X205.00 Y165.00 Z10.00
Move Away: 5.00mm
>>> do_homing_move  X205.00 Y165.00 Z10.00
...(X, 5.00, [50.00])
<<< do_homing_move  X205.00 Y165.00 Z10.00
Re-bump: -10.00mm
>>> do_homing_move  X205.00 Y165.00 Z10.00
...(X, -10.00, 25.00)
<<< do_homing_move  X205.00 Y165.00 Z10.00
>>> set_axis_is_at_home(X)
Axis X home_offset = 0.00 position_shift = 0.00
> home_offset[X] = 0.00
  current_position= X0.00 Y165.00 Z10.00 : 
<<< set_axis_is_at_home(X)
  current_position= X0.00 Y165.00 Z10.00 : sync_plan_position
  current_position= X0.00 Y165.00 Z10.00 : > AFTER set_axis_is_at_home
<<< homeaxis(X)
>>> homeaxis(Y)
Home Fast: -495.00mm
>>> do_homing_move  X0.00 Y165.00 Z10.00
...(Y, -495.00, [50.00])
echo:busy: processing
echo:busy: processing
<<< do_homing_move  X0.00 Y165.00 Z10.00
Move Away: 5.00mm
>>> do_homing_move  X0.00 Y165.00 Z10.00
...(Y, 5.00, [50.00])
<<< do_homing_move  X0.00 Y165.00 Z10.00
Re-bump: -10.00mm
>>> do_homing_move  X0.00 Y165.00 Z10.00
...(Y, -10.00, 25.00)
<<< do_homing_move  X0.00 Y165.00 Z10.00
>>> set_axis_is_at_home(Y)
Axis Y home_offset = 0.00 position_shift = 0.00
> home_offset[Y] = 0.00
  current_position= X0.00 Y0.00 Z10.00 : 
<<< set_axis_is_at_home(Y)
  current_position= X0.00 Y0.00 Z10.00 : sync_plan_position
  current_position= X0.00 Y0.00 Z10.00 : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
>>> home_z_safely  X0.00 Y0.00 Z10.00
  current_position= X0.00 Y0.00 Z10.00 : sync_plan_position
  destination= X205.00 Y165.00 Z0.97 : home_z_safely
>>> do_blocking_move_to  X0.00 Y0.00 Z10.00
>  X205.00 Y165.00 Z10.00
echo:busy: processing
<<< do_blocking_move_to  X205.00 Y165.00 Z10.00
>>> homeaxis(Z)
  current_position= X205.00 Y165.00 Z10.00 : Probe::set_deployed
deploy: 1
Probe::do_z_raise(10.00)
>>> do_blocking_move_to  X205.00 Y165.00 Z10.00
>  X205.00 Y165.00 Z10.97
<<< do_blocking_move_to  X205.00 Y165.00 Z10.97
>>> do_blocking_move_to  X205.00 Y165.00 Z10.97
>  X205.00 Y165.00 Z10.97
<<< do_blocking_move_to  X205.00 Y165.00 Z10.97
Home Fast: -600.00mm
>>> do_homing_move  X205.00 Y165.00 Z10.97
...(Z, -600.00, [4.00])
<<< do_homing_move  X205.00 Y165.00 Z10.97
Move Away: 5.00mm
>>> do_homing_move  X205.00 Y165.00 Z10.97
...(Z, 5.00, 4.00)
echo:busy: processing
<<< do_homing_move  X205.00 Y165.00 Z10.97
Re-bump: -10.00mm
>>> do_homing_move  X205.00 Y165.00 Z10.97
...(Z, -10.00, 2.00)
<<< do_homing_move  X205.00 Y165.00 Z10.97
>>> set_axis_is_at_home(Z)
*** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
> probe.offset.z = -0.97
Axis Z home_offset = 0.00 position_shift = 0.00
> home_offset[Z] = 0.00
  current_position= X205.00 Y165.00 Z0.97 : 
<<< set_axis_is_at_home(Z)
  current_position= X205.00 Y165.00 Z0.97 : sync_plan_position
  current_position= X205.00 Y165.00 Z0.97 : > AFTER set_axis_is_at_home
  current_position= X205.00 Y165.00 Z0.97 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to  X205.00 Y165.00 Z0.97
>  X205.00 Y165.00 Z0.97
<<< do_blocking_move_to  X205.00 Y165.00 Z0.97
<<< homeaxis(Z)
<<< home_z_safely  X205.00 Y165.00 Z0.97
  current_position= X205.00 Y165.00 Z0.97 : sync_plan_position
X:205.00 Y:165.00 Z:0.97 E:0.00 Count A:29600B:3200 Z:390
<<< G28  X205.00 Y165.00 Z0.97
ok

```
[MarlinConfig.zip](https://github.com/MarlinFirmware/Marlin/files/8006700/MarlinConfig.zip)

Boldie, 2022-02-04:
I also did some working around with the settings, in the configuration I posted above, I had USE_PROBE_FOR_Z_HOMING turned on. As I read somewhere else, only one of both can be active. I checked the behaviour on resetting it back to
```
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

// Force the use of the probe for Z-axis homing
// #define USE_PROBE_FOR_Z_HOMING
```
which does not work at all. This feature is not very well documented and also read the stuff on the marvin site, but did not figure out what this shall do.

```
Marlin bugfix-2.0.x
echo: Last Updated: 2022-02-04 | Author: (none, default config)
echo: Compiled: Feb  5 2022
echo: Free Memory: 13999  PlannerBufferBytes: 1600
echo:V86 stored settings retrieved (674 bytes; crc 18605)
echo:No SD card
M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,DETAIL
ok
>>> G28  X0.00 Y0.00 Z0.00
Machine Type: CoreCartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X-40.00 Y0.00 Z0.00 (Left-Center & Same Z as Nozzle)
Auto Bed Leveling: BILINEAR (disabled)

Raise Z (before homing) by 10.00
>>> do_blocking_move_to  X0.00 Y0.00 Z0.00
>  X0.00 Y0.00 Z10.00
echo:busy: processing
<<< do_blocking_move_to  X0.00 Y0.00 Z10.00
>>> homeaxis(X)
Home Fast: -495.00mm
>>> do_homing_move  X0.00 Y0.00 Z10.00
...(X, -495.00, [50.00])
echo:busy: processing
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z10.00
Move Away: 5.00mm
>>> do_homing_move  X0.00 Y0.00 Z10.00
...(X, 5.00, [50.00])
<<< do_homing_move  X0.00 Y0.00 Z10.00
Re-bump: -10.00mm
>>> do_homing_move  X0.00 Y0.00 Z10.00
...(X, -10.00, 25.00)
<<< do_homing_move  X0.00 Y0.00 Z10.00
>>> set_axis_is_at_home(X)
Axis X home_offset = 0.00 position_shift = 0.00
> home_offset[X] = 0.00
  current_position= X0.00 Y0.00 Z10.00 : 
<<< set_axis_is_at_home(X)
  current_position= X0.00 Y0.00 Z10.00 : sync_plan_position
  current_position= X0.00 Y0.00 Z10.00 : > AFTER set_axis_is_at_home
<<< homeaxis(X)
>>> homeaxis(Y)
Home Fast: -495.00mm
>>> do_homing_move  X0.00 Y0.00 Z10.00
...(Y, -495.00, [50.00])
echo:busy: processing
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z10.00
Move Away: 5.00mm
>>> do_homing_move  X0.00 Y0.00 Z10.00
...(Y, 5.00, [50.00])
<<< do_homing_move  X0.00 Y0.00 Z10.00
Re-bump: -10.00mm
>>> do_homing_move  X0.00 Y0.00 Z10.00
...(Y, -10.00, 25.00)
<<< do_homing_move  X0.00 Y0.00 Z10.00
>>> set_axis_is_at_home(Y)
Axis Y home_offset = 0.00 position_shift = 0.00
> home_offset[Y] = 0.00
  current_position= X0.00 Y0.00 Z10.00 : 
<<< set_axis_is_at_home(Y)
  current_position= X0.00 Y0.00 Z10.00 : sync_plan_position
  current_position= X0.00 Y0.00 Z10.00 : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
>>> home_z_safely  X0.00 Y0.00 Z10.00
  current_position= X0.00 Y0.00 Z10.00 : sync_plan_position
  destination= X205.00 Y165.00 Z0.00 : home_z_safely
>>> do_blocking_move_to  X0.00 Y0.00 Z10.00
>  X205.00 Y165.00 Z10.00
echo:busy: processing
<<< do_blocking_move_to  X205.00 Y165.00 Z10.00
>>> homeaxis(Z)
  current_position= X205.00 Y165.00 Z10.00 : Probe::set_deployed
deploy: 1
Probe::do_z_raise(10.00)
>>> do_blocking_move_to  X205.00 Y165.00 Z10.00
>  X205.00 Y165.00 Z10.00
<<< do_blocking_move_to  X205.00 Y165.00 Z10.00
>>> do_blocking_move_to  X205.00 Y165.00 Z10.00
>  X205.00 Y165.00 Z10.00
<<< do_blocking_move_to  X205.00 Y165.00 Z10.00
Home Fast: -600.00mm
>>> do_homing_move  X205.00 Y165.00 Z10.00
...(Z, -600.00, [4.00])
<<< do_homing_move  X205.00 Y165.00 Z10.00
Move Away: 5.00mm
>>> do_homing_move  X205.00 Y165.00 Z10.00
...(Z, 5.00, 4.00)
echo:busy: processing
<<< do_homing_move  X205.00 Y165.00 Z10.00
Re-bump: -10.00mm
>>> do_homing_move  X205.00 Y165.00 Z10.00
...(Z, -10.00, 2.00)
<<< do_homing_move  X205.00 Y165.00 Z10.00
>>> set_axis_is_at_home(Z)
*** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
> probe.offset.z = 0.00
Axis Z home_offset = 0.00 position_shift = 0.00
> home_offset[Z] = 0.00
  current_position= X205.00 Y165.00 Z0.00 : 
<<< set_axis_is_at_home(Z)
  current_position= X205.00 Y165.00 Z0.00 : sync_plan_position
  current_position= X205.00 Y165.00 Z0.00 : > AFTER set_axis_is_at_home
  current_position= X205.00 Y165.00 Z0.00 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to  X205.00 Y165.00 Z0.00
>  X205.00 Y165.00 Z0.00
<<< do_blocking_move_to  X205.00 Y165.00 Z0.00
<<< homeaxis(Z)
<<< home_z_safely  X205.00 Y165.00 Z0.00
  current_position= X205.00 Y165.00 Z0.00 : sync_plan_position
X:205.00 Y:165.00 Z:0.00 E:0.00 Count A:29600B:3200 Z:0
<<< G28  X205.00 Y165.00 Z0.00
```

Boldie, 2022-02-04:
Digging a little bit further, I found M119, which tells me the Z is beeing triggered, so I have to find out what is the problem here.: 
```
M119
Reporting endstop status
x_min: open
y_min: open
z_min: TRIGGERED
filament: open
ok
```

The interesting thing is, if I trigger the sensor once after bootup (power cycling, reset on board does keep it working) it shows the correct state. Ok I have to track this down tomorrow.

ellensp, 2022-02-04:
use build environment chitu_v5_gpio_init

From the .ini file 

#
# Some Chitu V5 boards have a problem with GPIO init.
# Use this target if G28 or G29 are always failing.
#
[env:chitu_v5_gpio_init]
extends       = env:chitu_f103
build_flags   = ${env:chitu_f103.build_flags} -DCHITU_V5_Z_MIN_BUGFIX

descipher, 2022-02-05:
PA14 is a JTAG pin, sounds like some boards have a bootloader that may be messing with it.
https://www.st.com/resource/en/reference_manual/CD00171190-.pdf page 177

Boldie, 2022-02-05:
I have used the chitu_v5_gpio_init already. I am not really convinced the build system compiles the right sources. I have altered buildroot/share/PlatformIO/variants/marlin_CHITU_F103/wirish/boards_setup.cpp to add some stuff and nothing happens. This is also the case if I add something which should never compile. I also can't see the file in the compile list after doing a clean.

Boldie, 2022-02-05:
Finally I got it work, but does not know how to add this properly to Marlin. I just added some toggling of Pull-Up on the beginning of setup function. Looks weird but works for the moment:

```
void setup() {
  GPIOA->BSRR = (0U << PA14);
  GPIOA->BSRR = (1U << PA14);

  #ifdef FASTIO_INIT
    FASTIO_INIT();
  #endif
```

nikkisstuff, 2022-02-08:
I have the same problem on a bigtreetech octopus v1.0 using a inductive sensor. Ive gone back to the last version.

geemo66, 2022-03-30:
> Finally I got it work, but does not know how to add this properly to Marlin. I just added some toggling of Pull-Up on the beginning of setup function. Looks weird but works for the moment:
>
> ```
> void [...]

Where did you add this to Marlin?

closed 2022-06-10 by github-actions[bot]