MarlinFirmware/Marlin #23680: [BUG] meatpack not available despite being enabled (Artillery Ruby)

gudvinr, 2022-02-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I enabled meatpack support with `#define MEATPACK_ON_SERIAL_PORT_2` but `M115` returns `Cap:MEATPACK:0`.  
If I enable meatpack plugin in OctoPrint it can't connect to printer until I disable plugin.

Relevant sections of config file (I didn't change these by myself. It is taken from configuration example for Artillery Ruby board):
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
#define BAUDRATE_2 115200   // Enable to override BAUDRATE
```

### Expected behavior

I expect meatpack to be enabled for virtual serial port on USB connection and be able to connect to printer using OctoPrint with meatpack plugin.

### Actual behavior

Meatpack doesn't seem to be enabled and OctoPrint stops connecting when meatpack plugin is enabled.

### Steps to Reproduce

1. Compile Marlin with meatpack support
2. Turn on printer
3. Connect to printer using OctoPrint

### Version of Marlin Firmware

2.0.9.3

### Printer model

Artillery Genius Pro

### Electronics

Stock Artillery Ruby board

### Add-ons

Alwinner A13 SBC with OctoPrint installed

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

I connected filament runout sensor to endstop connector for Z-axis and modified `pins_ARTILLERY_RUBY.h` to accommodate for those changes.

[genius-pro.zip](https://github.com/MarlinFirmware/Marlin/files/8008514/genius-pro.zip)

ellensp, 2022-02-05:
M115 only responds with the capabilities of the port you are connected with.
If you did not issue M115 from serial_port_2 it will report that meatpack is off

gudvinr, 2022-02-06:
> M115 only responds with the capabilities of the port you are connected with. If you did not issue M115 from serial_port_2 it will report that meatpack is off

Thanks. I tried enabling meatpack on serial 1 and it did help.

However, this raises quite an obvious question. If printer uses 2 (and more more) serial ports (one for tft and another for USB) how can you tell which one it is? It is kinda a good idea either distinguish them by name (like `SERIAL_TFT` and `SERIAL_USB` instead of just `SERIAL_1` and `SERIAL_2`) or at least introduce mandatory comment or something like that.

closed 2022-02-06 by gudvinr