MarlinFirmware/Marlin #23681: [BUG] T1 does not use second extruder

CortezGonzalas, 2022-02-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

2 extruders with different nozzle sizes
Gcode calls T1, the second nozzle preheats, but extruding comes from the first extruder. I can manually use EE2 from the LCD screen to move the second extruder which works fine. Also if the print switches between T0 and T1 (multimaterial print), it works fine. Printing with T0 only works as expected.

Gcode Snippet:
T1
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
G0 Z3
M190 S35 ;wait for bed to get to 35C to level
M140 S60 ;set bed temp
G28 ;home
G29 ;level bed
M190 S60
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:340
;LAYER:0
M107
G0 F3600 X100.091 Y74.875 Z0.2 ;starting print...

### Expected behavior

I expect T1 to use the extruder #2.

### Actual behavior

T1 uses extruder #1

### Steps to Reproduce

Add T1 to beginning of gcode.
Ensure no other Tool calls are in gcode
Print

### Version of Marlin Firmware

02000903

### Printer model

Anet A8 (sort of)

### Electronics

Ramps 1.4/Aduino Mega

### Add-ons

Second extruder/hotend

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8009012/Configuration.zip)

DerAndere1, 2022-02-06:
Have you tried to put the `T1` (select tool 1) command after the `G28` (auto home) command ? I guess G28 changes to tool 0 by default. I'm not sure whether it is supposed to change back to T1 after homing is done, or not.

ellensp, 2022-02-06:
I was thinking after g29
But yes, please move T1 down the list till it starts to work. so can tell what gcode breaks it

DerAndere1, 2022-02-06:
Also, please describe the actual hardware setup of your extruders and hotends. Maybe you have to enable `DUAL_X_CARRIAGE`, `SWITCHING_NOZZLE`, `PARKING_EXTRUDER`, `SWITCHING_TOOLHEAD` or `HOTEND_OFFSET_X`... offsets, depending on your hardware.

CortezGonzalas, 2022-02-06:
Hardware setup: 2 nozzles on a single carriage. I have tried both having hotend_offset_x=30 in Marlin and hotend_offset_x=0 (commented out) and then using Cura to control hotend offset. 

Moving T1 after the G29 works and it extrudes from the second extruder. Also, removing G29 completely with T1 at the beginning also works. So it looks like G29 resets the tool to T0.

G28 does not seem to affect the active tool.

ellensp, 2022-02-07:
digging around the code in src/gcode/bedlevel/abl/G29.cpp
I see
`    TERN_(HAS_MULTI_HOTEND, if (active_extruder) tool_change(0));`
Which changes to tool 0

ellensp, 2022-02-07:
give this a try
```DIFF
diff --git a/Marlin/src/gcode/bedlevel/abl/G29.cpp b/Marlin/src/gcode/bedlevel/abl/G29.cpp
index fddca22c32..2219e8bbc9 100644
--- a/Marlin/src/gcode/bedlevel/abl/G29.cpp
+++ b/Marlin/src/gcode/bedlevel/abl/G29.cpp
@@ -262,6 +262,9 @@ G29_TYPE GcodeSuite::G29() {
    * On the initial G29 fetch command parameters.
    */
   if (!g29_in_progress) {
+    #if HAS_MULTI_HOTEND
+      uint8_t save_active_extruder = active_extruder;
+    #endif
 
     TERN_(HAS_MULTI_HOTEND, if (active_extruder) tool_change(0));
 
@@ -452,6 +455,10 @@ G29_TYPE GcodeSuite::G29() {
       }
     #endif // AUTO_BED_LEVELING_BILINEAR
 
+    #if HAS_MULTI_HOTEND
+      if (save_active_extruder) tool_change(save_active_extruder);
+    #endif
+
   } // !g29_in_progress
 
   #if ENABLED(PROBE_MANUALLY)

```

CortezGonzalas, 2022-02-07:
Seems to have done the trick. It's working as it should now. Thanks!

ellensp, 2022-02-07:
ok ill turn it into a PR shortly, fix it in main code.

DerAndere1, 2022-02-07:
Interesting that your solution works. Out of intuition I would have tried to restore the extruder later in abl/G29.cpp, just before this section

```
}
  #endif // !PROBE_MANUALLY
  //
  // G29 Finishing Code
  //
```

ellensp, 2022-02-11:
This fix has been merged. Closing

closed 2022-02-11 by ellensp