MarlinFirmware/Marlin #23683: [BUG] Probing does not work with home offset

davkvi, 2022-02-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

#### About the printer

**Version:** [Ender3v2-422-BLTouch-20220202.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20220202/Ender3v2-422-BLTouch-20220202.bin)
**Settings:** [Imgur album](https://imgur.com/a/VmAirX8)
**Mods:**
- Changed [mounting plate](https://i.imgur.com/dbFfV7F.png) for hotend to the ender 3 pro
- Changed [Fanduct](https://www.thingiverse.com/thing:3721746)

[image]

The red line illustrates where the probe and hotend starts.
The line is aprox 8.5mm off. It might be related to chainging the mounting plate. That does not explain why it levels all the way in the front of the bed. Have tried various different fixes. But the problem does not resolve. Seems to be an upstream issue

## Please refrence this [bug report](https://github.com/mriscoc/Ender3V2S1/issues/29)

### Bug Timeline

New.

### Expected behavior

Probe following the green line and start in center

### Actual behavior

Probes following the red line and starts at yellow dot.

### Steps to Reproduce

Please refrence imgur album for settings.

But set probe offsets, and then set home offset. 

### Version of Marlin Firmware

bugfix

### Printer model

Ender 3 V2

### Electronics

Creality 4.2.2

### Add-ons

Swaped mounting plate

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

Firmware changes is from this https://github.com/mriscoc/Ender3V2S1/releases/tag/20220202

ellensp, 2022-02-06:
These files are not marlin. they are a fork. If you have issues with a fork you must open issues on their repo.

closed 2022-02-06 by ellensp

davkvi, 2022-02-06:
> These files are not marlin. they are a fork. If you have issues with a fork you must open issues on their repo.

I belive it to be an upstream issue with marlin and not the fork.

ellensp, 2022-02-06:
Then you need to reproduce this on real marlin and provide the configs. since mriscoc breaks licensing and does not does not provide config or source there is no way to look into this as is.

mriscoc, 2022-02-06:
> Then you need to reproduce this on real marlin and provide the configs. since mriscoc breaks licensing and **does not provide config or source** there is no way to look into this as is.

Sorry ellensp but you are in a mistake about config and sources, the config files were always in the repository and also most of the source. It was a small piece of the code not published, but all were functional; Please don't repeat that the others say without checking for yourself.

tititopher68-dev, 2022-02-06:
@ellensp 
You are completely right.  The famous proui.a library still does not respect the license (it has been analyzed as one would for a virus by looking for code analogies and bingo!).  It is a problem only related to its version using this library.  He modified part of the probe.h code by bypassing rules using home X/Y Offset and Probing margin.  So nothing to do with Marlin but only with its fork!