MarlinFirmware/Marlin #23684: Stealthchop TMC2209 skipping steps,loss of torque on X Y Z E motori making bad noice

SamppaD78, 2022-02-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi. Just updated to 2.9.3 Marlin with pre defined configs for SKR mini E3 2.0 on a heavily modified Ender 3 pro. Before update stepper motors have been running in stealthchop mode with current set to XYZE: 520,520,650,600
Stelthchop was enabled but hybridtreshhold was disabled in old firmware.Motors has been quite,cool and no skipping steps. Hoewer this changed after update,. compiled Marlin and I modified few things:
Set tmc 2209 current to XYZE to 550,550,650,600. 
Enabled hybridtreshhold XYZE  100mm/s for X and Y  Z set to 3mm/s E set to 30mm/s 
Reset main board and screen,upload firmware and after this I  restore settings,load settings,save settings. Confirmed this with pronterface M500->M503
After firmware update,total loss of positioning torque when stelthchop is enabled,all motors sounds like they failed, mechanically,internally,bad noise when motors are moving,at the same time skipping steps.
Recompiled,re upload..same thing. Tested bugfix,same issue. Increase current to XYZ 580,580,670..no luck
Noticed this when I disable stelthchop future from printer,motors sounds normal. Hoewer if the future was already active making noise and you disable future after this happens it looks like it is not actually 100% disabled.Issue persist even after reboot. What i mean that it seems that tmc2209 are running partially in stelthchop mode? Motor sound is not normal and some steps skipping acuress. Work around this is to re upload firmware with enabled stelthchop,and then disable stelthchop from the printer screen and save settings. After this when printing motors sounds normal,positioning torque on X,Y,Z is strong.
Also noticed that  if  hybridtreshhold disabled marlin will not recompile without it enabled. 
Do anyone know is there any changes made by Marlin on stelthchop,hybridtreshold and tmc2209 code?
Asking this because my printer is normally quite without stelthchop,when this is enabled it sounds like totally busted.
More about printer.
Ender 3 pro,Dragon all metal hotend,BLV linear reail kits X Y Z, dual motors on Z axis,reinforced frame,SKR mini e3 2.0 ,BTT tft 35 screen.Waiting to resolve this issue so direct drive can be fitted to this puppy.

### Bug Timeline

New bug,started after 2.9.3 update

### Expected behavior

Expected that when steltchop is active motors works normally,no skipping,movement is smooth and quite.
Also hoping that hybridtreshold can be dissabled even if the stelthchop is enabled,at least this was posiblle in 2.9 version

### Actual behavior

Motor sounds like DC10 lifting off at the same time braking appart. Mechanicle noise,vibrations and skipping steps. Week positioning torque on all axis.

### Steps to Reproduce

First compile Marlin using configs for skr mini e3 2.0
When stelthchop XYZE,hybridtreshold enabled issue is present. Motors making mechanical like broken noise,loss of torque on all axis. Disabling stelthchop after this issue has been noticed on printer will not actually dissable this future 100% in firmware. Motors will still run partially is stelthchop,torque is better after disabling this on printer but still motor noise is not normal and steps are skipped. Only after re upload of firmware and disabling stelthchop future on screen before starting to home axis or print resolve this problem

### Version of Marlin Firmware

2.0.9.3

### Printer model

Crality Ender 3 Pro

### Electronics

SKR mini E3 2.0

### Add-ons

no add-ons

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

My config has  : Linear advance off,classic jerk off,squarewave stepping on, arc support on,monitor driver status,tmc debug on,s curve on,junction deviation on, stelthchop XYZE,hybridtreshold enabled. TMC current set to XYZ 550,550,650(dual Z axis) 600 for extruder. Otherwise it is stock 2.9.3 Marlin.
Now running old firmwire no degug enabled so i cant provide you with M111 S247 info
[config.zip](https://github.com/MarlinFirmware/Marlin/files/8009920/config.zip)

Arakon, 2022-02-06:
I ran 2.0.9.3 with both an SKR Mini E3 V2.0 and V3.0 with Stealthchop always enabled and have no such issues, so this is not a general problem with the boards, at least. However, I have Hybrid Treshold disabled. Maybe disable that and see if that is the culprit?

SamppaD78, 2022-02-06:
Hi. this is what i suspect,hoewer Marlin will not compile when hybrid treshold is disabled. I get errors like not defined XYZ hybrid treshold values. On previus marlin version i had stelthchop enabled but hybrid treshold was disabled and havent had any issues . Latest Marlin doesn't allow stelthcop without hybrid treshold enabled. I dont know why is this the case as i normally don need printing  speeds over 80mm/s

oliver-eifler, 2022-02-06:
the tmc_init function always needs a [axis]_HYBRID_THRESHOLD value, even if it is not used due
disabled HYBRID_THRESHOLD - seemed to need some code cleanup ;)
to disable hybrid_mode/switching between stealthchop and spreadcycle according to speed
just disable
//#define HYBRID_THRESHOLD
and let all the [axis]_HYBRID_THRESHOLD enabled
[image]

and/or set the [axis]_HYBRID_THRESHOLD to crazy high values (i.e. 999999) 
that never could be reached and switching between stealthchop and spreadcycle will never happen regardless of HYBRID_THRESHOLD enabled

SamppaD78, 2022-02-19:
Hi. Finally found some time to do some testings  with Marlin . I can confirm that Olivers advice, to disable hybrid threshold  and leave all the axis  hybrid threshold enabled solved my problem.. Printer is working fine, I haven't noticed any torque loss on motors but i will continue testing this. Now i come across a different issue  regarding a hi temperature thermistor not displaying  room temperature correctly. It is a dyze design t-d500 500C thermistor with thermistor table number 66 in marlin. I got some help from community as to just ignore room temp issue or to make a custom temp table as marlin might have some issue with this thermistor preset values. I would expect that manufacturer resistance thermistor table values would be the same as Marlin preset as this thermistor was supported from version 1.1.0 RC7 but I cant verify this. I don't mind few degree fluctuation in temp readings but this sensor reports at stable 21C temperature,readings about 28-36C . It looks like this sensor by design have some difficulties reading temps below 40C,so i enabled #define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 5
#define MILLISECONDS_PREHEAT_TIME 5000.

closed 2022-04-21 by thinkyhead