MarlinFirmware/Marlin #23698: [BUG] MKS Robin Nano V3 no PSU Control with PWC possible

Liriel-666, 2022-02-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When Setup the MKS Robin Nano v3 with a PWC Unit. 

I must activate the PSU Control in  the Config

#define PSU_CONTROL
//#define PSU_NAME "Power Supply"

#if ENABLED(PSU_CONTROL)
  #define MKS_PWC                 // Using the MKS PWC add-on
  //#define PS_OFF_CONFIRM          // Confirm dialog when power off
  //#define PS_OFF_SOUND            // Beep 1s when power off
  #define PSU_ACTIVE_STATE LOW      // Set 'LOW' for ATX, 'HIGH' for X-Box

  //#define PSU_DEFAULT_OFF         // Keep power off until enabled directly with M80
  //#define PSU_POWERUP_DELAY 250   // (ms) Delay for the PSU to warm up to full power

  //#define PSU_POWERUP_GCODE  "M355 S1"  // G-code to run after power-on (e.g., case light on)
  //#define PSU_POWEROFF_GCODE "M355 S0"  // G-code to run before power-off (e.g., case light off)

But then this message comes at compile
```
> Executing task: C:\Users\Laen\.platformio\penv\Scripts\platformio.exe run <

Processing mks_robin_nano_v3 (board: marlin_STM32F407VGT6_CCM; platform: ststm32@~12.1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
In file included from c:\drucker\marlin-bugfix-2.0.x--08.02.22\marlin\src\pins\stm32f4/pins_MKS_ROBIN_NANO_V3.h:72,
                 from c:\drucker\marlin-bugfix-2.0.x--08.02.22\marlin\src\pins\pins.h:669,
                 from c:\drucker\marlin-bugfix-2.0.x--08.02.22\marlin\src\inc\marlinconfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\drucker\marlin-bugfix-2.0.x--08.02.22\marlin\src\pins\stm32f4/pins_MKS_ROBIN_NANO_V3_common.h:169:8: error: #error "PSU_CONTROL is incompatible with MKS_PWC plus TFT_LVGL_UI."
  169 |       #error "PSU_CONTROL is incompatible with MKS_PWC plus TFT_LVGL_UI."
      |        ^~~~~
CalledProcessError: Command '"C:\Users\Laen\.platformio\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-g++.exe" 
-D__MARLIN_FIRMWARE__ -DNDEBUG -DHAL_STM32 -DUSBCON -DUSBD_USE_CDC -DTIM_IRQ_PRIO=13 -DPIN_WIRE_SCL=PB6 -DPIN_WIRE_SDA=PB7 -DHAL_PCD_MODULE_ENABLED -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\Users\Laen\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 179:
    env.SConscript(item, exports="env")
  File "C:\Users\Laen\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:       
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Laen\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:       
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\drucker\Marlin-bugfix-2.0.x--08.02.22\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 247:
    apply_features_config()
  File "C:\drucker\Marlin-bugfix-2.0.x--08.02.22\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 133:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\Laen\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\drucker\Marlin-bugfix-2.0.x--08.02.22\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 216:
    load_marlin_features()
  File "C:\drucker\Marlin-bugfix-2.0.x--08.02.22\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 204:
    define_list = run_preprocessor(env)
  File "C:\drucker\Marlin-bugfix-2.0.x--08.02.22\buildroot\share\PlatformIO\scripts\preprocessor.py", line 42:       
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Users\Laen\.platformio\python3\lib\subprocess.py", line 424:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\Laen\.platformio\python3\lib\subprocess.py", line 528:
    raise CalledProcessError(retcode, process.args,
============================================ [FAILED] Took 2.08 seconds ============================================

Environment        Status    Duration
-----------------  --------  ------------
mks_robin_nano_v3  FAILED    00:00:02.079
======================================= 1 failed, 0 succeeded in 00:00:02.079 =======================================Der Terminalprozess "C:\Users\Laen\.platformio\penv\Scripts\platformio.exe 'run'" wurde mit folgendem Exitcode beendet: 1.
```

I think ther eis a error because in the Release 2.0.9.3 it works

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/8026169/configs.zip)

### Bug Timeline

new

### Version of Marlin Firmware

bugfix 2.0.x from 8.4.2022

### Printer model

Anet A8

### Electronics

MKS Robin Nano V3

### Add-ons

MKS PWC

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

Liriel-666, 2022-02-08:
I think it is  the Code in the pins_MKS_ROBIN_NANO_V3_common.h

line 169
 #error "PSU_CONTROL is incompatible with MKS_PWC plus TFT_LVGL_UI."

that prevent it but not exist in 2.9.0.3

but psu control must be on for pwc use

Liriel-666, 2022-04-06:
Nothing fixed until today

thisiskeithb, 2022-04-06:
> I think ther eis a error because in the Release 2.0.9.3 it works

How? `PSU_CONTROL` was silently undefined if `TFT_LVGL_UI` was enabled in `2.0.9.3`:

https://github.com/MarlinFirmware/Marlin/blob/f471eab1a2834c4e65477d978ea9f0349542b302/Marlin/src/pins/stm32f1/pins_MKS_ROBIN_NANO_common.h#L117-L131

I wouldn't call this a bug, but a feature request.

Liriel-666, 2022-04-06:
That is a Bug or someone miss understand it and disable it In The bugfix.
Without lvgl and Power Control no pwc but All these three thing must work together.

thisiskeithb, 2022-04-06:
The link above is straight from 2.0.9.3. It *did not* work as you think it does because Marlin was silently disabling enabled features while compiling.

Liriel-666, 2022-04-06:
Strange pwc and lvl works fine

Rockman18, 2022-05-25:
In 2.0.9.3, it works perfectly because the MKS_PWC variable is only necessary to set the SUICIDE_PIN and SUICIDE_PIN_STATE and, to avoid other unwanted behaviour, MKS_PWC and PSU_CONTROL are disabled.

PSU_CONTROL is not important for LVGL_UI so the undef is here to avoid stange behaviour.
MKS_PWC is mandatory in LVGL_UI to set SUICIDE_PIN and SUICIDE_PIN_STATE and only for this so the feature can be disabled after that.

For LVGL_UI, the #define MKS_PWC can be outside  of #if ENABLED(PSU_CONTROL) but to keep the logic define PSU_CONTROL before is not idiot even if it is not necessary.

Liriel-666, 2023-01-28:
ok then marlin after 2.0.9.3 is useless because lvgl_ui and pwc are not useable together.

EvilGremlin, 2023-01-28:
This mess is because lvgl_ui implements it's own functions instead of using existing ones, and no one wanted to fix it yet. People generally prefer color_ui adn classic_ui. If someone will step up to fix and maintain it (at least for a bit) we will be very grateful.

Liriel-666, 2023-01-28:
and with color ui the pwc module can be used?

EvilGremlin, 2023-01-28:
You just setup it as regular PSU_CONTROL stuff.

closed 2024-01-10 by thinkyhead