MarlinFirmware/Marlin #23699: [BUG] on LVL Temp Display Overlapping when 2 Extruder are set

Liriel-666, 2022-02-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

On Marlin 2.0.9.3 and bugfix 2.0.3 the Temperatur Display in the LVL Main Screen are Overlapping.

Setup is 1 Heatbed and 2 Hotend Sensors.

I think its because the Positions are Change!
[20220208_175509](https://user-images.githubusercontent.com/10562572/153046663-0dec90ca-5d43-4fa8-8af2-41ca74a56f4f.jpg)

I think ist because the order was change.

Old normal was:
Heatbed  - Hotend 1 - Hotend 2

now
Hotend 1 - Hotend 2 - Heatbed

Best would be Change order back oder change the space between Hotend 22 and Heatbed

### Bug Timeline

first for me first appear in 2.0.9.3

### Version of Marlin Firmware

bugfix 2.0.x from 8.4.2022

### Printer model

Anet A8

### Electronics

MKS Robin Nano V3

### Add-ons

Dual Hotend and Nozzle

### Bed Leveling

ABL Bilinear mesh

ellensp, 2022-02-11:
Himm after a merry chase around the code, I found the little blighter

in Marlin/src/lcd/extui/mks_ui/draw_ready_print.cpp line 199 

```CPP
      buttonBedstate = lv_big_button_create(scr, "F:/bmp_bed_state.bin", " ", TERN(HAS_MULTI_HOTEND, 271, 210), ICON_POS_Y, event_handler, ID_INFO_BED);
```

Change 271  to 340

```CPP
      buttonBedstate = lv_big_button_create(scr, "F:/bmp_bed_state.bin", " ", TERN(HAS_MULTI_HOTEND, 340, 210), ICON_POS_Y, event_handler, ID_INFO_BED);
```

This works for me, give it a try

Liriel-666, 2022-02-11:
yes with that change it is perfect

thisiskeithb, 2022-02-11:
https://github.com/MarlinFirmware/Marlin/pull/23718 has been merged.

closed 2022-02-11 by thisiskeithb