MarlinFirmware/Marlin #23700: [BUG] Filament Change "Please wait for prompt" does not go away DWIN_CREALITY_LCD_JYERSUI

severandynasty, 2022-02-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When using the Filament Change menu to load filament, the printer will not allow you to purge more filament using the "Continue" prompt.  The UI becomes unresponsive and the printer must be power cycled to restore function.  Bug is present in the 2.0.9.3 release and current bugfix branch.  Filament load works correctly when DWIN set is changed to the MarlinUI.

User @doubleO00 also reported this issue.

### Bug Timeline

Appears to exist for some time, issue was also reported under issue 22772

### Expected behavior

Filament should load to initial load length as defined in FILAMENT_CHANGE_FAST_LOAD_LENGTH then prompt user to purge more if needed or exit.  If purge more (Continue) is selected the printer should purge an additional amount each time the user presses Continue until the process is exited.

### Actual behavior

Filament load runs the FILAMENT_CHANGE_FAST_LOAD_LENGTH then prompts user to purge more or exit.  If purge more is selected the menu freezes at Filament Change "Please wait for prompt".  If users exits several other menu items like Disable Steppers no longer work.  Printer must be power cycled to restore operation.

### Steps to Reproduce

1. Power on printer
2. Select Prepare -> Change Filament -> Load Filament
3. Select appropriate temperature from Preheat menu when prompted then press Continue.  Printer heats to required temperature.
4. Printer loads filament and prompts user to Purge More (Continue) or Exit (Cancel)
5. User selects Continue, printer freezes at "Filament Load, Please wait for prompt
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8026322/Configuration.zip)
"

### Version of Marlin Firmware

2.0.9.3

### Printer model

Creality Ender 3 v2

### Electronics

Stock 4.2.2

### Add-ons

BLtouch 3.1, Filament Runout Sensor (microswitch)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

tome9111991, 2022-02-11:
> @severandynasty modify dwin.cpp in /src/lcd/e3v2/jyersui/, line 5001 by :
> 
> ```
> case PAUSE_MESSAGE_OPTION: 
>         pause_menu_response = PAUSE_RESPONSE_WAIT_FOR;
>         CrealityDWIN.Popup_Handler(PurgeMore);
>         break;
> ```
> 
> or use this version 2.0.2i here : [Jyers#1694 (comment)](https://github.com/Jyers/Marlin/discussions/1694#discussion-3749966)

Maybe Anyone feels free to create a PR with this fix?

ellensp, 2022-02-11:
@tome9111991  Has this been confirmed to work?

tome9111991, 2022-02-11:
> @tome9111991 Has this been confirmed to work?

I have only tried @tititopher68-dev version.
I think fix is included and there  some other fixes it works nice.

I dont know maybe the can add all the fixes here?

Dids, 2022-03-30:
I can confirm that the fix works. Tried it on my own fork and it no longer gets stuck on the prompt, allowing user input (confirm/cancel).

htro, 2022-04-29:
> @severandynasty modify dwin.cpp in /src/lcd/e3v2/jyersui/, line 5001 by :
>
> ```
> case PAUSE_MESSAGE_OPTION:
>         pause_menu_response = PAUSE_RESPONSE_WAIT_FOR; [...]

This seems to be only partial fix, it hangs the firmware if purge more functionality is used during filament runout situation. IIRC it also hanged the firmware if the purge more is used multiple times during normal filament change.

ellensp, 2022-07-07:
JyersUI has been removed from Marlin.  See https://github.com/MarlinFirmware/Marlin/pull/24459 for details.

closed 2022-07-07 by ellensp