MarlinFirmware/Marlin #23702: [FR] Setting different print acceleration values for X and Y

Smitty-Werbenjagermenjensen, 2022-02-09:
### Are you looking for hardware support?

Ender 3, SKR Mini e3 V3.0 and Creality 4.2.2. 

### Describe the feature you want

My bed is a glass bed. As a result, it must use a relatively low acceleration, due to the bed's mass and weight, in order to achieve good print quality.  

My Bowden print head on the other hand is very light and can accelerate very fast without degrading print quality. 

My ideal print acceleration values: 
Y-axis : 300 mm/s^2
X-axis : 700 mm/s^2

I was able to set the maximum acceleration values in the firmware to these values. So this is what I have in code `#define DEFAULT_MAX_ACCELERATION      { 700, 300, 100, 10000 }`. I set those values as maximum is because I don't want my printer to use any faster acceleration, since those acceleration values result in the best print quality. 

Currently, I can only set one print acceleration value, P. 

I set it as follows : `#define DEFAULT_ACCELERATION          300    // X, Y, Z and E acceleration for printing moves`

I set it to 300 since this is the lowest acceleration. 

Or can this be achieved by setting P to say 700 mm/s^2? 

With P = 700 mm/s^2, would X use 700 mm/s^2 as defined by the max value? And then Y would use 300 mm/s^2 since it is the maximum value?

ManuelMcLure, 2022-02-09:
Just set the maximum acceleration for Y with either `M201` or `DEFAULT_MAX_ACCELERATION`.

ManuelMcLure, 2022-02-09:
If you set acceleration to 700 it will still not exceed the 300 acceleration you set for Y.

Smitty-Werbenjagermenjensen, 2022-02-10:
Okay, so my question is, how can I do the following: 

After I set the max acceleration as X = 700 and Y = 300, how can I make the X-axis use 700 acceleration and Y-axis 300 acceleration during a print, when there is only one printing acceleration, "P"? I can only set one value for P.

thisiskeithb, 2022-02-10:
> how can I do the following

Your slicer should manage that for you:
[image]

ManuelMcLure, 2022-02-10:
The axis accelerations do not work completely independently - they are always related to each other. Marlin always calculates what the acceleration of the _total move_ is, which might have both X and Y components, and will limit the _total move_ depending on what components of the move are in each dimension.

For example, suppose you move 100 in X and 10 in Y. If you limit the X acceleration to 700mm/sec^2, Y will only accelerate at 70mm/sec^2, so in this case the Y acceleration never reaches the limit you set for Y acceleration. If you instead move 10mm in X and 100mm in Y, the Y acceleration limit comes into effect and X will only accelerate at 30mm/sec^2. Marlin will never accelerate any axis higher than the maximum per-axis acceleration you specify, even if the total move acceleration is higher.

Smitty-Werbenjagermenjensen, 2022-02-11:
> The axis accelerations do not work completely independently - they are always related to each other. Marlin always calculates what the acceleration of the _total move_ is, which might have both X [...]

I understand that and thank you for your reply. 

But what is the purpose and role of the print acceleration P?

Smitty-Werbenjagermenjensen, 2022-02-11:
> > how can I do the following
> 
> Your slicer should manage that for you: [image]

Thanks. It seems like Cura does not support that unfortunately. It only sets the print acceleration, which is just P. This goes back to my original question. And this makes sense that Cura (and other slicers) can't do it, because M204 only allows setting one value, P, just like in firmware. I cannot adjust the print acceleration for the individual axes. 

The values shown in your screenshot are the maximum values I have referred to. Those can be set per axis, as the slicer is showing. All that the slicer is doing is running M201 commands and setting the X,Y,Z, and E max acceleration. I have set my max acceleration in firmware and that is not an issue. 

So, in conclusion, I know that slicers (through the M201) command and the firmware can be used to set the maximum acceleration. However, my concern is that there is a printing accleration, P, in firmware. This can be set with M204. To my understanding and based on information from the forums, this P acceleration is what is used during printing/extruding operations. My problem and cause of this post is that P is only one value and not per axis.

ManuelMcLure, 2022-02-11:
> I understand that and thank you for your reply.
> 
> But what is the purpose and role of the print acceleration P?

If P is less than the per-axis acceleration then the printer will only accelerate at that speed. Usually the per-axis accelerations are higher than the P value so in that case P is the acceleration used for the total move.

Smitty-Werbenjagermenjensen, 2022-02-11:
Okay, so then I need to set P at or above the largest limit? 

So, if I have a maximum X = 700 and Y=300, and set P = 700, then the printer will use 700 for X and 300 for Y?

I know you mentioned how Marlin chooses and uses acceleration. But that aside, it will use 700 for X and 300 for Y in that situation?

ellensp, 2022-02-11:
You cannot accelerate at different rates, the axis are locked to the lowest rate for any moves that use multiple axis

Smitty-Werbenjagermenjensen, 2022-02-11:
> You cannot accelerate at different rates, the axis are locked to the lowest rate for any moves that use multiple axis Otherwise the axis get out of sink position wise.

Goodness. Everyone is saying something different. 

So what happens if I set P = 700, but my Y-max is 300 and my X-max is 700?

ellensp, 2022-02-11:
then it is set to 300

Everyone is confused, because this request is not founded in 3d printer reality.

When you request a move say X50Y50 Marlin ensures that it arrives at X50Y50 at the same time. If it didnt instead of a diagonal line you would draw (in your example of 700 for Y and 300 for X) either and arc or a inverted L type shape.   Depending on the implmentation

ellensp, 2022-02-11:
It would seem to me you want some of  motion control platform.. where all axis are totally independent.
This is not marlin.

Smitty-Werbenjagermenjensen, 2022-02-11:
Which axes are interdependent like that?

I assume just the X and Y are coupled? The E and Z are independent for the most part?

ManuelMcLure, 2022-02-11:
All axes are coupled. If E wasn't coupled to X and Y you'd get incorrect extrusion. If a move includes a Z component it is also coupled to X and Y. Marlin figures out the trigonometry for the move and limits the acceleration to the most restrictive axis involved in the move as I explained. At shallow angles from direct X motion the X acceleration limit is the limiting factor because Y acceleration will never reach the Y limit, as you add more and more Y component into the move at some point the Y acceleration will become the limiting factor.
The Z acceleration/jerk limits can actually cause prints using bed leveling on very warped beds to slow down, although this is uncommon.

Smitty-Werbenjagermenjensen, 2022-02-11:
> All axes are coupled. If E wasn't coupled to X and Y you'd get incorrect extrusion. If a move includes a Z component it is also coupled to X and Y. Marlin figures out the trigonometry for the move [...]

Thanks. yes that makes complete sense. 

But ellensp mentioned that if P=700, then both X and Y will actually move at P=300. This is because the Y-Max is 300. And so even though X=700 is the max, the x axis will still only accelerate at 300. 

The Z-axis has a max acceleration of 100 and my extruder has a max acceleration of 10,000. These will not be effected in any way by P, right? Like I understand the motion and how it plans the movements, and therefore everything is intertwined. If X is only moving at 200 then E will only move at some other value. 

So by "coupled", I meant how X and Y will be set to the same value of 300. But I was thinking that Z and E are more independent and more consequential, in terms of the math and movement planning.

ManuelMcLure, 2022-02-11:
If you're moving at a 45 degree angle, so the X and Y move the same amount, then yes, the two axes will accelerate at 300mm/sec^2. But if you move 100 in X and 50 in Y then Y will accelerate at 300mm/sec^s and X will accelerate at 600mm/sec^2 (still not the 700mm/sec^2 that is the maximum for that axis).

Smitty-Werbenjagermenjensen, 2022-02-11:
Okay I see. 

You really know alot. For my setup, what value of P should I set? 

I mentioned how my X-axis can go 700 with good print quality. And my Y-axis can go 300 with good print quality. 

With that said, what would you set as P?

ManuelMcLure, 2022-02-11:
That's better covered in a different forum. We've really moved out of the "reporting bugs" function of Github. Please join the Marlin discord for support issues.

closed 2022-02-12 by ellensp