MarlinFirmware/Marlin #23709: [BUG] (Marlin 2.0.9.3)

Haiang, 2022-02-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I made my own 3D printer board design for myself, I got some errors like the image below that I don't know where and how these errors appear.
I'm currently using Marlin 2.0.9.3
And below is my Flatform IO core config, am I doing something wrong?
[env:VN_S63_V1_0]
platform = ststm32@<6.2.0
board = genericSTM32F103RC
platform_packages = tool-stm32duino
build_flags = !python Marlin/src/HAL/STM32F1/build_flags.py
   ${common.build_flags} -DDEBUG_LEVEL=0 -std=gnu++14 -DHAVE_SW_SERIAL -DSS_TIMER=4
build_unflags = -std=gnu++11
src_filter = ${common.default_src_filter} +<src/HAL/STM32F1>
lib_deps = ${common.lib_deps}
lib_ignore = SPI
custom_marlin.NEOPIXEL_LED = Adafruit NeoPixel=https://github.com/CommandoreBombardiero/Adafruit_NeoPixel

### Bug Timeline

New

### Expected behavior

Need help

### Actual behavior

Checked the Bug fix version but the error is still there

### Version of Marlin Firmware

2.0.9.3

### Printer model

Ender

### Electronics

STM32F103RC

### Your Slicer

Simplify3D

### Host Software

Cura

### Additional information & file uploads

[image]
[image]
[image]
[image]
[image]

ellensp, 2022-02-11:
main issue is probably the very  old platform = ststm32@<6.2.0

try aiming for ststm32@~12.1

ellensp, 2022-02-11:
Any defined but not used warning are safe to ignore, but any others should be looked into

Haiang, 2022-02-11:
> main issue is probably the very old platform = ststm32@<6.2.0
> 
> try aiming for ststm32@~12.1

If I change to ststm32@~12.1
then get this error
Marlin\src\HAL\shared\../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or directory
    30 | #include <libmaple/gpio.h>
       | ^~~~~~~~~~~~~~~~
compilation terminated.

ellensp, 2022-02-11:
check the current ini/stm32f1-maple.ini
[common_stm32f1]
platform          = ststm32@~12.1
board_build.core  = maple       <===

Haiang, 2022-02-11:
I still can't handle this problem
Here is my file and configuration, please help me see it
https://drive.google.com/file/d/1bGjovjugCApikgzdExem79ygyLJImcbt/view?usp=sharing

ellensp, 2022-02-12:
Updating for a non maple environment 

[env:VN_S63_V1_0]
extends                     = common_STM32F103RC_variant
build_flags                 = ${common_STM32F103RC_variant.build_flags}
                              -DTIMER_SERVO=TIM5
board_build.offset          = 0x7000     <===  change this to suit your bootloader
lib_ignore    =  SoftwareSerialM
monitor_speed     = 115200

closed 2022-04-25 by github-actions[bot]