MarlinFirmware/Marlin #23710: Marlin GPLv3 License violation

VincentJGeisler, 2022-02-10:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Creality makes printers using modified marlin firmware. They provide "source code" of an older version, but it is not the modified source code they use to produce their binaries.  They have been warned, requested and fully understand that their actions violate the terms of the Marlin-GPLv3 license.

### Bug Timeline

new

### Expected behavior

Creality modified source code should provide identical binaries in function. It does not. they have left out the critical modifications that allow the touch screen to work. 

### Actual behavior

downloaded the 1.70.1 source code from creality, compiled. The compiled binary was half the size of the precompiled binary that the printer ships with, and the touchscreen did not work. 

### Steps to Reproduce

download the source provided by creality https://drive.google.com/drive/folders/16596EvFWv9C9ClbavmHPc4J3MJ4_N3h_
compile
compare binaries beteeen precompiled and source compiled. Notice the difference in size. Creality did not provide all their source modifications. 

### Version of Marlin Firmware

1.70.1

### Printer model

creality ender 5 plus

### Electronics

stock

### Add-ons

stock

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

### Additional information & file uploads

creality slicer

ellensp, 2022-02-10:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-02-10 by ellensp

VincentJGeisler, 2022-02-10:
Cool, do that means marlin is abandoning its rights under gpl3? By closing this ticket you provided evidence of acceptance of licensing violation.  Under legal doctrine if you don’t defend your copyleft you lose it. 

Act accordingly

ellensp, 2022-02-10:
See  https://forums.creality3dofficial.com/download/ender-series/ender-5-plus/ 

So you very wrong in this accusation.

ellensp, 2022-02-10:
I repeat: 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

VincentJGeisler, 2022-02-10:
Have you actually tried compiling it? 
The binary is half the size of the pre compiled version and results in a non functional touch screen. 

Ergo they are in licensing violation. 
I am very right because:law school, protect it or lose it.

Regards

ellensp, 2022-02-10:
You are incorrect.

Compiled, just now from their provided source code 
350K Feb 10 22:08 Marlin.ino.hex
Provided hex file
350K Sep  4  2019 'Ender-5PlusBLTouch_0904_V1.70.1 BL.hex'

Is it byte perfect, no, you would have to use the exact same compiler version they used for that. 

And compiled firmware works fine with provided DWIN_SET  (you always have to match the DWNI_SET with the correct MArlin code) 

Yes I just tested it. 
Please stop wasting our time

thinkyhead, 2022-02-10:
> I am very right because:law school, protect it or lose it.

That applies to trademarks, but not to copyright.