MarlinFirmware/Marlin #23713: [BUG] In Marlin language files

roel8032, 2022-02-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

[BUG] Marlin language files. On the LCD to move Z-axis in 0.025mm steps.
The language system does not support translate this to other languages.
Variable not in language_en.h:

 LSTR MSG_MOVE_N_MM                      = _UxGT("Move %smm");
  LSTR MSG_MOVE_01MM                      = _UxGT("Move 0.1mm");
  LSTR MSG_MOVE_1MM                       = _UxGT("Move 1mm");
  LSTR MSG_MOVE_10MM                      = _UxGT("Move 10mm");
  LSTR MSG_MOVE_100MM                     = _UxGT("Move 100mm");
  LSTR MSG_MOVE_0001IN                    = _UxGT("Move 0.001in");
  LSTR MSG_MOVE_001IN                     = _UxGT("Move 0.01in");
  LSTR MSG_MOVE_01IN                      = _UxGT("Move 0.1in");
  LSTR MSG_MOVE_1IN                       = _UxGT("Move 1.0in");

So you can not translate it in your own language file.

[image]

So you can not translate it in your language file.

### Bug Timeline

Old

### Expected behavior

Possible to translate

### Actual behavior

Wrong word

### Version of Marlin Firmware

Any version.

### Printer model

Multiple

### Electronics

Stock boards and more.

ellensp, 2022-02-10:
I just tried this in latest bugfix.
It works as expected
That line uses LSTR MSG_MOVE_N_MM

I deliberately changed this to  = _UxGT("XXrplaats %smm");  in Marlin/src/lcd/language/language_nl.h

And sure enough it changed on the LCD
[image]

ellensp, 2022-02-10:
I can only presume you have a old language_nl.h that does not have the line
LSTR MSG_MOVE_N_MM                      = _UxGT("Verplaats %smm");

ellensp, 2022-02-10:
This line was added to marlin in 2.0.8 on wards

closed 2022-02-10 by ellensp

roel8032, 2022-02-10:
Confirm. Working now
Added the line to language_nl.h

 LSTR MSG_MOVING                          = _UxGT("Verplaatsen...");
  LSTR MSG_FREE_XY                         = _UxGT("Vrij XY");
  LSTR MSG_MOVE_X                          = _UxGT("Verplaats X");
  LSTR MSG_MOVE_Y                          = _UxGT("Verplaats Y");
  LSTR MSG_MOVE_Z                          = _UxGT("Verplaats Z");
  LSTR MSG_MOVE_E                          = _UxGT("Extruder");
  LSTR MSG_MOVE_EN                         = _UxGT("Extruder *");
  LSTR MSG_MOVE_Z_DIST                     = _UxGT("Verplaats %smm");
  LSTR MSG_MOVE_N_MM                       = _UxGT("Verplaats %smm");  <====================
  LSTR MSG_MOVE_01MM                       = _UxGT("Verplaats 0.1mm");
  LSTR MSG_MOVE_1MM                        = _UxGT("Verplaats 1mm");
  LSTR MSG_MOVE_10MM                       = _UxGT("Verplaats 10mm");