MarlinFirmware/Marlin #23716: [BUG] (bug summary)

doucettom, 2022-02-10:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I tried to recompile a fresh Marlin 2.0.9.3
https://github.com/MarlinFirmware/Marlin/issues/20824

### Bug Timeline

Old, since 2.0.7.2

### Expected behavior

Not rebooting

### Actual behavior

Reboots every few seconds

### Version of Marlin Firmware

2.0.9.3

### Printer model

MP Select Mini V2

### Electronics

stock

### Add-ons

None

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/8044979/Marlin.zip)

doucettom, 2022-02-11:
Duplicate of #20824

doucettom, 2022-03-04:
This is reproducible on 3 different MP Select Mini v2's.

tablatronix, 2022-03-05:
Yeah Same here anyone figure out what the issue is before I stick a debugger on this, that issue was closed with no help

doucettom, 2022-03-06:
https://www.reddit.com/r/MPSelectMiniOwners/comments/l08gyq/marlin_2072_on_mp_select_mini/hzjkoue/?context=3

thisiskeithb, 2022-03-06:
Looks like the bug was introduced in https://github.com/MarlinFirmware/Marlin/pull/19693.

If commenting out the following works, then I can put in a PR to remove it from the M200 pins file:

https://github.com/MarlinFirmware/Marlin/blob/323ac9488ce0562611d0144ab8ba700d272c00c7/Marlin/src/pins/stm32f1/pins_MALYAN_M200.h#L36-L37

thisiskeithb, 2022-03-07:
Anyone have a V1? Since this setting is inherited, it may not apply to all versions.

doucettom, 2022-03-07:
I don't but the v2 has its own file that refers to v1's file.

thisiskeithb, 2022-03-07:
> I don't but the v2 has its own file that refers to v1's file.

That’s why I’m asking. The setting is inherited from the V1’s pins file, so I don’t want to disable it on the V1 if it’s not affected.

tablatronix, 2022-03-07:
Right, but disabling a watchdog is not a fix for anything

julian-weinert, 2022-07-13:
May I bump this? I think I'm running into the same issue, but am not entirely sure.
I'm new to Marlin and all this, but my LCD module properly boots, but the motor drivers stay enabled and the main board doesn't seem to report parameters to the LCD module.

thinkyhead, 2023-02-16:
Please try disabling the display and other features one-by-one, in case any particular feature is contributing to the problem.

dzid26, 2023-02-27:
> Anyone have a V1? Since this setting is inherited, it may not apply to all versions.

Just tried Merlin for the first time using this [configuration](https://github.com/MarlinFirmware/Configurations/tree/068346365dbc5633823b6e4afae9f8d5dadd4dd0/config/examples/Malyan/M200) and getting bootloop on V1 as well.  Just before it resets I hear the extruder stepper vibration, but it's not rotating. 
In between the resets I can shortly control the printer via LCD (except the extruder). 
Also, Octoprint is able to establish a connection:
```Recv:  T:41.56 /0.00 B:20.44 /0.00 @:0 B@:0
Trying port /dev/ttyACM0, baudrate 500000
Connecting to port /dev/ttyACM0, baudrate 500000
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0x6fd8b370, open=True>(port='/dev/ttyACM0', baudrate=500000, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Detecting serial connection" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin bugfix-2.1.x (Feb 26 2023 22:22:05) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Malyan M200 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

---------------------------------------------------------------------------
Your printer's firmware is a development build of Marlin (build date
20230226). It might be more unstable than a release version and should be
kept up-to-date.

Learn more at https://faq.octoprint.org/warning-firmware-development
---------------------------------------------------------------------------
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:0
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_POS:0
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:RUNOUT:0
Recv: Cap:Z_PROBE:0
Recv: Cap:LEVELING_DATA:0
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:HOST_ACTION_COMMANDS:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:SDCARD:1
Recv: Cap:MULTI_VOLUME:0
Recv: Cap:REPEAT:0
Recv: Cap:SD_WRITE:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:1
Recv: Cap:LFN_WRITE:0
Recv: Cap:CUSTOM_FIRMWARE_UPLOAD:0
Recv: Cap:EXTENDED_M20:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:0
Recv: Cap:BABYSTEPPING:0
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: Cap:COOLER_TEMPERATURE:0
Recv: Cap:MEATPACK:0
Recv: Cap:CONFIG_EXPORT:0
Recv: ok
Send: M21
Recv: echo:SD card ok
Recv: ok
Send: M155 S2
Recv: ok
Send: M20 L
Recv: Begin file list
Recv: PUMPHO~1.GCO 7446555 Pump holder.gcode
Recv: End file list
Recv: ok
Recv:  T:41.13 /0.00 B:20.77 /0.00 @:0 B@:0
Recv:  T:40.97 /0.00 B:20.71 /0.00 @:0 B@:0
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3997
Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Changing monitoring state from "Operational" to "Offline after error"
Connection closed, closing down monitor
```

closed 2024-09-24 by github-actions[bot]